ftctechnh/ftc_app #1: Default API version for compiling is set to 21

calebsander, 2015-06-04:
FtcRobotController/build.gradle instructs Gradle to try to build the app using Android SDK 21, which doesn't make sense given that it is designed for SDK 19 and SDK 22 is the newest release version.

acharraggi, 2015-06-05:
I noticed that too, I wondered why I had to download API 21 when the "targetSdkVersion 19" . I think the 21 is coming from compileSdkVersion = 'Google Inc.:Google APIs:21'.  I let Android Studio build it that way, it works, but we shouldn't need that. International teams may not be able to use the ZTE Speed phones so their set up could be different, but I think it Android 4.4 is still recommended.

I think we'll want to set compileSdkVersion to 19 based on:
http://stackoverflow.com/questions/26694108/what-is-the-difference-between-compilesdkversion-and-targetsdkversion

I tried this and was able to get it to run. I changed compileSdkVersion = 'Google Inc.:Google APIs:19' and also set minSdkVersion 19.
File at: https://github.com/acharraggi/ftc_app/blob/master/FtcRobotController/build.gradle

Seems to work ok, I've got the driver and robot apps running on the ZTE phones (but with no other hardware yet).

pshah123, 2015-06-05:
Can you make a pull request so everyone gets an updated copy? Thanks!

agileconnor, 2015-06-07:
@acharraggi While I agree with you on the topic of European users having differences in hardware, I disagree on your judgement that we should downgrade to API 19. The target SDK has no effect on the software or the phones that it runs on. All that is saying is that this is the API we tested it the most and where it is guaranteed to work. This is used in the Android app store to determine which phone cannot run the software / what version the app is optimized for.
By using the latest API we can get access to newer features only available in that version (ie Better multithreading). Android has automatic backwards compatibility so even if the phone is running API 19 it can still use the application.

**Sources**
* https://developer.android.com/training/basics/firstapp/creating-project.html
* I'm an Android developer

JacobAMason, 2015-06-07:
I have to agree with @agileconnor; leaving the compileSdkVersion at 21 will allow the app to build on devices running Lolipop (Android 5.0).  I have built the app for Lolipop on my OnePlus and I didn't have any build errors.  I might need to change the targetSdkVersion to do a true test of sdk21/Lolipop for this app (maybe?).
In addition, the [.iml file here](https://github.com/acharraggi/ftc_app/commit/ee642e2e0a9407c3078270bc6d2ee7466965872d ) probably shouldn't have been included in the repository to begin with, since I *believe* that is a file regenerated at build.

acharraggi, 2015-06-07:
If this was a general purpose Android app, then the highest API level possible would be best. And perhaps ftctech should leave it as is since this is an example we can all use and some will deploy to newer versions.

But once you copy this app, you are going to customize it for your robot and it's only going to run on your robot controller phone and that will have a specific API level. That app will not run on an earlier API or a later API, it's only running on one phone.  

If you leave compileSdkVersion at 21 but you're phone's API is 19, you can possibly include calls to features not supported. eg. Sensor.TYPE_HEART_RATE requires API level 20, it's flagged as 'field requires API level 20 (current min is 19)' when the min is 19 and compileSdkVersion is 21. You're allowed to build it even though it won't do what you want. Change compileSdkVersion to 19 however and you get a 'cannot resolve symbol' error. Also, Android Studio helps with auto-complete based on the compileSdkVersion so you don't even see TYPE_HEART_RATE as an option to pick from.

Why not let the compiler flag newer stuff that won't work on your phone?

I'm not sure we need to change compileSdkVersion at source. But I do think each team should set compileSdkVersion to match their phones after they copy the software.  I think now I would recommend "build instructions" in the README mention some recommendation or suggestions. And that may come as part of User manual that the README says is forthcoming.

acharraggi, 2015-06-07:
@JacobAMason: I agree, FtcRobotController.iml could be removed from the repository. 

I confirmed that FtcRobotController.iml will be rebuilt if you start with a clean directory and that IML deleted and then open that directory with "Open an existing Android Studio project". 

The build process populates the API version in the IML file. If you also change build.gradle prior to the build to set compileSdkVersion to 19 you get <orderEntry type="jdk" jdkName="Android 4.4.2 Google APIs" jdkType="Android SDK" /> in the IML file, instead of the 5.0.1 reference.

ftclhsrobotics, 2015-08-05:
The *FtcRobotController.iml* is an IntelliJ module file and should **not** be part of the repository

closed 2015-08-06 by calebsander

ftclhsrobotics, 2015-08-15:
You closed the issue but you did not remove the **FtcRobotController.iml**

Do you feel is still necessary?

calebsander, 2015-08-15:
That seems like a different issue. I was originally only referring to FtcRobotController/build.gradle, although I agree removing FtcRobotController.iml is a good idea.

ftclhsrobotics, 2015-08-15:
Yeah...I think the topic was hijacked :-)