ftctechnh/ftc_app #3: GyroSensor.getRotation() returns what exactly?

pshah123, 2015-06-05:
I'm asking because it was left unclear, does this return the rate of rotation or the rotation angle? And is the angle a quaternion, Euler angle, etc. ?
Also, this is all assuming this refers to a GyroSensor that is one of the new sensors, not the Built in or legacy ones, right?

calebsander, 2015-06-05:
Well, a gyro sensor measures speed of rotation, and there is only one number returned, so I assume it is either degrees per second (like with Xander's libraries for the HiTechnic sensors) or radians per second.

pshah123, 2015-06-05:
True, thanks! I was thinking along those lines but some newer sensors for other devices sent out angles after doing the math themselves so I was unsure.

closed 2015-06-05 by pshah123

calebsander, 2015-06-10:
Actually, after looking at the decompiled code for ModernRoboticsNxtGyroSensor, its return value is just a straight read from the analog port, or (degrees per second) + ~620. See how [Xander](https://github.com/botbench/rdpartyrobotcdr/blob/master/drivers/hitechnic-gyro.h#L72) managed this in RobotC.