ftctechnh/ftc_app #5: Can we use the ZTE phone sensors? Is there a better way than what I tried?

acharraggi, 2015-06-10:
I created an OpMode to return Android sensor values, in this case related to device orientation. 

Do we know if FTC teams going to be allowed to use the phone sensors?

Is there a way to do this with RobotCore or some other API in that's been included? 
I didn't see anything, but only looked briefly. 

To do it I had to save the application context from FtcRobotControllerActivity and make use of the Android Sensor classes. The following is a link to GitHub compare to show what I changed and what I added to make it work: https://github.com/acharraggi/my_ftc_app/compare/v0.1...v0.2

It seems to work OK. I'm just not sure if it's safe to do (thread safe or any other kind of safe: such as proper un-registering of the sensor listeners in all cases). I don't have the new hardware yet to test using both together.

What do you think?

JacobAMason, 2015-06-10:
While phone sensors would be awesome to use, I don't see us being allowed to do that since we aren't being forced to buy a particular phone. Therefore some phones would perform better than others, giving some teams an unfair advantage just because they have more money to spend on it. I'm definitely interested in what the official rules will be regarding what phone we have to use, too, for this specific reason.

acharraggi, 2015-06-11:
The FIRST press release hints that sensors are allowed: http://www.businesswire.com/news/home/20150311006143/en/FIRST%C2%AE-Announces-%E2%80%98Game-Changing%E2%80%99-Technology-Platform-Worldwide-Student#.VXjs1EbNvZW

*"Moving to a communication and control system based on Android and Snapdragon will give FIRST Tech Challenge students powerful benefits during their build and competition seasons, including:
    High speed processing and advanced sensors already built into phones and tablets based on Snapdragon; "*

Teams with money have always had advantages: more parts, custom parts, more competitions that they can participate in by travelling, etc. Just part of the game, and true for FLL, FTC and FRC. FIRST could say we have a fixed list of parts to work from, but I think they intentionally leave it pretty open. I think this year they're just trying to make the transition easier by having a standard phone. I think the year after this one you'll be free to choose your phone, perhaps subject to some restrictions like being able to run the driver station app.

JacobAMason, 2015-06-11:
Oooh cool. I'm looking forward to the phone's built-in accelerometer and things like that.
Yeah, I suppose that's true; the open kit of parts does make the competition and the whole design process considerably more interesting. Maybe we'll eventually be customize the app a bit more, too.

acharraggi, 2015-06-11:
And how about using the camera? One of our kids wants to send the robot's camera image to the driver station (might help with lining up on scoring elements). He's also experimenting with edge detection using the camera. He sent a screen shot showing a Frisbee on a playing field and how an edge detection routine had converted the image to a bitmap of just the edges. The Frisbee clearly showing as a rough circle.with the mat edges as lines. Of course now he has the challenge of interpreting the bitmap. But still, it's the start of robot vision processing.

ftctechnh, 2015-06-13:
Hi Folks,

Please post questions in the FTC Tech forum:

http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology

acharraggi, 2015-06-14:
Ok but I'm still waiting for my ID to be allowed to post there. I do have a
specific issue which is: does ftc-app support Android Sensors? If so, how?

acharraggi, 2015-06-18:
Thanks to GearTicks on the FTC Tech forum who pointed out *hardwareMap.appContext*. I was able to remove the custom code I added to save context and use that instead and it works. Now I guess we wait for the game rules to see if using the phone sensors is allowed.

closed 2015-06-18 by acharraggi

FTC12457, 2017-10-08:
Wait so can the phone sensors be used?

acharraggi, 2017-10-08:
Yes. Phone sensors are not prohibited by rule RE11. The camera on the phone is effectively a sensor.  They want you to use Vuforia and the camera on the phone to analyze the pictographs.