ftctechnh/ftc_app #7: Joystick support in driver station

pbrier, 2015-06-17:
I cannot seem to get the Logitech F310 joystick working on many (european) Android phones.
With the switch in "X" position it is seen as a USB device with VID=0x046d and PID=0xC21D, but no
driver is loaded. The `xpad.ko` driver is not loaded on many phones, and it seems to be required. 

Can the driver station app use direct USB communication to read the controllers? 
(I can see the USB data from the controller when I use the tincore App)

leoforney, 2015-07-31:
Try testing hitting the mode button and see if it turns green.

We had a similar issue and If you are using a ZTE Speed, our first phone had very bad issues detecting the controller, so we used one of our other phones and it worked. Also press the Start button + X.

ftctechnh, 2015-10-16:
Hi pbrier,

I am going to close this question out.  We would like to use this database to track known issues/bugs with the system.  

If you need assistance troubleshooting your robot, please visit the FTC Technology forum.  This issue tracking DB should be reserved for verified bugs/issues.

http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology

Thanks!

Tom

closed 2015-10-16 by ftctechnh

TheOutlaws, 2015-11-02:
And I am having troubles my team wants me to program the robot to go straight then turn and thiks is my first year doing android could I get a push in the right direction for this program