ftctechnh/ftc_app #10: Please use the FTC Technology forum to post questions and report issues regarding the FTC SDK

ftctechnh, 2015-07-21:
Hi Folks,

Please post all questions/issues in the FTC Technology forum:

http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology

The Technology forum is where you should visit if you have questions, comments or issues regarding the new platform (including the FTC SDK).

Thanks,

Tom

aaronraimist, 2015-07-25:
After the current issues are dealt with (by creating threads on the FTC forum) you can disable issues using the repo's [settings panel](https://github.com/ftctechnh/ftc_app/settings).

NoahAndrews, 2015-07-26:
Please don't disable the issue tracker. Just because you don't want to use it for support (which it wasn't even designed for) doesn't mean that you should remove the best way for us to report bugs and request new features.

aaronraimist, 2015-07-26:
@NoahAndrews I do agree that the GitHub issue tracker is a good bug reporting system but the above comment specifically says "Please post all questions/issues in the FTC Technology forum". If Tom wants people to use the forum then the issue tracker should not stay enabled. It would just cause confusion.

eukota, 2015-07-28:
A major difference is that anyone can submit reports on GitHub while the forum is constrained to those who can login. I have been waiting for my login to be approved for several months. I have used the contact form twice and still no luck.

So I cannot report issues.

AlecHub, 2015-10-13:
Thank you Tom & FTC for opening up this issue tracker: http://ftcforum.usfirst.org/showthread.php?4954-Bug-fix-priorities-and-release-planning&p=18506&viewfull=1#post18506

@eukota, good point.  I believe the repository owner (FTC) has the ability to control which GitHub users can create issue tickets or comment on issues tickets.

Alec

closed 2015-10-13 by ftctechnh