ftctechnh/ftc_app #11: Source .jars for libs

ftclhsrobotics, 2015-08-05:
Hello,

Would it be possible to get the source .jars for the jars in the libs directory
* FtcCommon-release.aar  (the App Activity classes and UI elements)
* RobotCore-release.aar (the framework base and utility classes)

Many of the students on the team are new to JAVA and just get introduced to the JAVA language and having the source code of the classes we use (like **OpMode**, **HardwareMap**, etc) would be very educational.  Some of the students are familiar to the Java SDK and were hoping to see an annotated source code, similar to how JAVA SDK is released, to better understand how the code is written.

And from the 'android' app side it was nice to have the source code for the various Activities used by the RobotController and DriverStation projects so that the kids could possible learn about Android software development elements.

Thanks for all that you have done!

techkid6, 2015-08-11:
+1 Because this should be obvious.  FTC wishes to inspire education, which means it *should* be well within their core values to open as much of their code as possible.  Considering the licensing on the aar's themselves, you would imagine putting up the source for at least FtcCommon (where the near entirety of the FTC app activities were moved to) would make sense.

I would like to add to this though.  Putting compressed archives in a git repository just makes things difficult.  Why not get together a Maven repository with all of these libraries and point the build.gradle file to them?  Gradle caches by default so building code offline (after the initial setup, which you have to do anyway to even get things going) is not at all a problem in this setup.

magneticflux-, 2015-09-14:
Yes, this would be highly useful. Being able to understand how the code works will allow teams to debug easier and write code faster. Putting the .jar files in the repo is also not the right way to do it. Use Maven, Gradle, or some other dependency management system.
I also think that the source for the driver station should be released, so teams can make their own modifications. Being able to use other methods besides joysticks would make teleop more interesting. I was envisioning touchscreen controls for viewing and setting motor powers, or setting an angle to turn to.

v3n, 2015-09-14:
+1

However, I suspect there might be some licensing issues there from the third parties that wrote those libraries. HOWEVER, the state of documentation right now without actually having source access is abysmal.

ksbdude, 2015-09-26:
I would like to have the ability to see the source code also. Having the source code would allow people to actually see what there code is doing and modify or even contribute to improving the library which could in the end benefit everybody. I think it would match what First is trying to strive for by releasing some or all of the source code.

WardBenjamin, 2015-09-26:
You actually can see the source code really easily (you just have to decompile it). Android Studio can actually do this for you if you right click on a class name of something in a library, then click "open declaration."

ksbdude, 2015-09-26:
Click on something and hit Ctrl + B to see code

magneticflux-, 2015-09-26:
@ksbdude @WardBenjamin Obviously. However, that shows a decompiled version that is missing names, comments, and basically anything a computer doesn't need to execute the code, but a human needs to understand the code.

ksbdude, 2015-09-26:
@magneticflux- Which is why releasing that would be helpful.

thee-l, 2015-10-01:
I absolutely agree. Any source code is going to be good.

strugee, 2015-10-07:
This has been raised before, in [this forum post][1]. From the little that FTC has said in that thread, seems like Qualcomm's at fault. AFAICT they're the ones hoarding the source.

 [1]: http://ftcforum.usfirst.org/showthread.php?4492-Open-Source-Please&p=16668#post16668

rgatkinson, 2016-09-04:
The 2.1 release includes sources for much of the SDK. These should reliably show up if you re-"import" the project into Android Studio. We hope to add more of the SDK sources in the future.

closed 2016-09-04 by rgatkinson