ftctechnh/ftc_app #15: Reading from a Motor Connected to Hi Technic Motor Controllers Fails

luciengaitskell, 2015-08-29:
If any read function is called, for example `arm.getCurrentPosition()` where `arm` is the hardware mapping for the motor connected to the Hi Technic Motor Controller or `controller.getMotorCurrentPosition(port)` were `controller` is the hardware mapping for the controller, it fails reading:
```User code threw an uncaught exception. Cannot read while in this mode: WRITE_ONLY from method:``` Then the method used.

calebsander, 2015-09-28:
The Legacy Motor Controllers cannot be in both read mode and write mode at once. You need to request a switch to READ_ONLY and then only run the read command when getMotorControllerDeviceMode() returns READ_ONLY.

luciengaitskell, 2015-09-30:
I found READ_WRITE works!

closed 2015-09-30 by luciengaitskell