ftctechnh/ftc_app #16: Incorrect compileSDKVersion

agileconnor, 2015-09-04:
complieSDKVersion is currently listed as "Google Inc.:Google APIs:19". This is incorrect version naming for default Android Studio setup. Should be listed "android-19".

calebsander, 2015-10-12:
It looks like it is often specified with simply the number, and the current way of doing it works fine.

hexafraction, 2015-10-12:
This does not compile after a clean import on my setup. The syntax `compileSdkVersion 19` works on multiple setups I have tried, including one using IntelliJ IDEA with an Android plugin rather than Android Studio directly.

RoboMWM, 2015-12-30:
I'm getting the same issue on build as well. Found [this](http://sagistech.blogspot.com/2011/06/android-sdk-error-unable-to-resolve.html) when searching around, see the latest comment.

rgatkinson, 2016-09-20:
I believe this is resolved in the current (2.2) release.

closed 2016-09-20 by rgatkinson