ftctechnh/ftc_app #23: Reading and Writing with an I2C device in a core device interface module

jspspike, 2015-09-21:
Can you please provide an example or explain how to use an i2c device like a gyro

strugee, 2015-10-07:
@jspspike your best bet currently is to search the forums.

jspspike, 2015-10-07:
They actually included an example program in the latest update so I'm all
good

closed 2015-10-09 by gearsincorg

eukota, 2015-10-09:
We have some more I2C code you can look at in our fork here: https://github.com/SwerveRobotics/ftc_app.

Specifically, look at the I2C Device Client Classes (https://github.com/SwerveRobotics/ftc_app/tree/master/SwerveRoboticsLibrary/src/main/java/org/swerverobotics/library/interfaces).

And our demo code which uses them to interface with a BNO055 IMU from AdaFruit: https://github.com/SwerveRobotics/ftc_app/blob/master/SwerveRoboticsLibrary/src/main/java/org/swerverobotics/library/examples/SynchIMUDemo.java