ftctechnh/ftc_app #30: Android Studio reads this as bad code.

Based-Brian10, 2015-09-29:
[image]
I have tried many code cleanup websites, but the issue the Studio is telling me is that I cannot resolve the symbol of OpModeRegister and I cannot resolve the method of .register. What is wrong and what must I do?

WardBenjamin, 2015-09-29:
I have seen this error before. You most likely haven't installed the correct Android SDK versions. Post a picture of your SDK Manager? You need to install SDK 21 and a few tools.

KevinEmery, 2015-09-29:
@WardBenjamin @Based-Brian10 You need to have the following SDK items installed according to the project configuration.

API 19 (SDK and Google APIs for ARM)
Build Tools v 21.1.2

This is contrary to the manual which instructs you to install API 21, but if you do this you will then need to install API 19 in order to run the project as it exists currently

MeRPG, 2015-09-30:
Brian, you may have not run a gradle build yet. Dependencies such as the ftc sdk won't work until you build your project.

Thank you,
- Jaxon Brown

ArchimedesPi, 2015-10-06:
Sync with Gradle (running a build will do this).

buddytex, 2015-10-06:
@Based-Brian10 If you are still having this issue; It appears that you have an error in your build.gradle,  take a look at that and if you can not figure it out please post a screencap here so that we may help you out. I wish you luck!

calebsander, 2015-10-12:
To second what others have been saying, we have seen this occur frequently before doing a Gradle Sync/Build. If you click "Build" (in the topmost menu) and then "Make Project", it should fix itself after a few seconds (wait until "Gradle Build Running" disappears from the bottom bar).

hexafraction, 2015-10-12:
I've experienced the case where a misconfigured firewall/content filter combined with JVM certificate validation with HTTPS caused this issue as Gradle tried to fetch artifacts from JCenter. Could you post the Gradle log, as seen by "Show logs in explorer", as a [gist](https://gist.github.com), and providing a link to that gist here?

Based-Brian10, 2015-10-13:
[image]
@buddytex @calebsander @MeRPG

Based-Brian10, 2015-10-13:
[image]
[image]
@WardBenjamin 
@KevinEmery

Based-Brian10, 2015-10-13:
@hexafraction I can't understand your question.

calebsander, 2015-10-13:
You are putting Java code in your gradle script, so it obviously is going to fail. Reset FtcRobotController/build.gradle to how it was and then put your MecanumDefaultCode in its own .java file.

Based-Brian10, 2015-10-13:
@calebsander how can I reset the code?

calebsander, 2015-10-13:
Download the file from this repository and replace your current version with it.

Based-Brian10, 2015-10-13:
@calebsander How do I put my Mecanum code in its own file?

MeRPG, 2015-10-13:
@Based-Brian10 @calebsander
Or you could use the 'git reset' command in git bash. https://git-scm.com/docs/git-reset

@Based-Brian10 Here is how you should go about creating a new op mode:

[image]
[image]
[image]

Also be careful. The code visible in one of the screenshots you sent is NOT java. It actually looks like a wierd mixture between Java and WPILib from FRC last season.

More information on how to set Android Studio/IntelliJ IDEA up for FTC can be found here:
https://raw.githubusercontent.com/ftctechnh/ftc_app/master/doc/tutorial/FTCTraining_Manual.pdf
Especially from pages 87-108.

Hope I could help.

ftctechnh, 2015-10-16:
Thanks to everyone who has assisted in this thread.  I am going to close this item out.  We would like to use this database to track verified issues/bugs with the systems.  Requests for assistance should be directed to the FTC Technology forum, or in this case to the Android Studio subforum:

http://ftcforum.usfirst.org/forumdisplay.php?159-Android-Studio

closed 2015-10-16 by ftctechnh

TheOutlaws, 2015-11-02:
And I am having troubles my team wants me to program the robot to go straight then turn and thiks is my first year doing android could I get a push in the right direction for this program

hexafraction, 2015-11-02:
@TheOutlaws This is not the correct venue for support. Please use the FTC official subforums for: [New FTC Technology](http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology) and [New FTC Technology > Android Studio](http://ftcforum.usfirst.org/forumdisplay.php?159-Android-Studio).