ftctechnh/ftc_app #38: Emergency Stop being called extraneously

nitrocalcite, 2015-10-13:
Hello, I am a member of a rookie FTC team.  With the latest release of the FTC app, we have started to encounter issues with the robot state "EMERGENCY_STOP".  It seems to call itself and shut down our robot unpredictably.  
          Oftentimes this happens when our robot is on the mountain and is jarred, even with nothing but the drive motors attached.  When the robot is level and we shake it, even bang on our securely-mounted electronics, the app continues to run.  If we take the phone off its mount and shake it/turn it upside down, the app continues to run.  
        However, if we attach the phone to the robot and tilt the robot like it was climbing the mountain, even a little shaking of the robot will cause the app to set the robot state to EMERGENCY_STOP.  This is a major problem because resetting the robot state requires restarting the app multiple times, something that could not happen during the competition.  
        We have tried to find the line of code which is triggering this condition.  However, there is only one "line" of functions, each calling one another, that involve this robot state.  This quickly leads us into a closed section of the FTC code which we cannot access.  
Any suggestions?

Thank you!

ftctechnh, 2015-10-13:
hi nitrocalcite - I think the behavior you are seeing might be due to a faulty connection.  The system has a  safeguard so that if the robot loses connection to any of its USB modules, it will do an e-stop.  

If an E-stop occurs, you should be able to recover (assuming the hardware is reconnected to the USB bus) remotely from the driver station (go to the main menu and select "Restart Robot".  

In order to avoid these errors please check your wiring and the mounting of your modules.   It might be that the cables are not secured and strain relieved or it could be that the jolt/jarring action is causing a brownout (knocking the power loose).   Pay special attention to the cable that connects the phone to the power module.  We've seen in the past some issues with this cable being loose (the one that connects the phone's USB micro port to the Power Module's USB mini port).  If they aren't secured properly, you might want to try to secure it.

Would you be able to pull a copy of the log file from your robot controller and post it over on the tech forum the next time you encounter a problem?  We can take a look and see what's causing the e-stop but it sounds like it might be a loose connection somewhere.

Tom

ftctechnh, 2015-10-16:
I am going to close out this question.  This question can be posed over at the FTC Technology forum.  The GitHub issue db should be reserved for reporting bugs or known system issues.  If you need assistance troubleshooting your system, please visit the FTC Technology Forum

http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology

closed 2015-10-16 by ftctechnh

nitrocalcite, 2015-10-16:
Thanks so much!  We forgot to test two wires which were the ones causing
the problem.  The USB cables don't hold in upside-down I guess.