ftctechnh/ftc_app #41: Potential Hardware Issue with early production Modern Robotics Power Modules

ftctechnh, 2015-10-16:
Modern Robotics has informed FIRST that they suspect that some of the early production Core Power Distribution Modules might have a hardware issue that can cause inconsistent behavior with the device's built-in USB hub.  Devices that are affected might make it difficult for the Android Robot Controller to detect USB modules (such as the motor controller, servo controller, legacy module, and/or device interface module) that are connected through the Power Distribution Module's USB hub.

The problem requires that the module be sent back to Modern Robotics to have a part replaced on the device.

FIRST has requested from Modern Robotics additional information to help identify potentially suspect Power Distribution Modules, but currently we do not have a serial number range that can help us identify the devices which might be affected by the issue.

If you are experiencing inconsistent USB connection issues that cannot be explained by problems with loose wiring or low battery voltage, then you should contact Modern Robotics for technical assistance to determine if you have a problematic device.

http://modernroboticsinc.com/contactus

ftclhsrobotics, 2015-10-17:
Modern Robotics needs to provide a serial number list so that we can quickly resolve this issue.

The fact that the problem manifests itself as "inconsistent USB connection issues" means that it **could** work through the _handy_ utitity program and still fail at other times.

Its quite easy once they identify the offending/misbehaving module any competent inventory system should be able to spit all the serial numbers of Core distribution modules that contain the bad parts

JochenFischer, 2015-10-23:
My team demonstrated their simple robot at the Mini Maker Faire in Charlotte two weeks ago. The robot was being used all day long and we lost connection numerous times.

As an engineer myself, I can say that it was not loose wiring nor low battery levels.

After trying to fix the connections all day long, I found that unplugging the USB cable from the Power and USB Distribution Module and plugging it in again always fixed the issue. It appears that the fault lies with the USB connection breaking down at some point.

techied, 2015-10-24:
Wow. Glad I stumbled upon this, we are losing connection randomly because of failures. Will definetly contact Modern Robotics.

mennesoft, 2015-10-25:
I can also verify that our team has one of the quirky core distribution
modules. Inconsistent is the best way to describe it. When you connect the
same set of modules and click Detect in the Settings of the Robot
Controller application, sometimes you'll get all the connected devices, but
sometimes not, and if you retry you'll often get different results even if
you don't touch the hardware or cables at all.

We have also experienced loss of connection, but it could be related to the
system trying to detect connected hardware at startup, and then getting
different results as explained above.

We'll contact Modern Robotics as well. I don't know our serial number off
hand.

Kevin Menningen
Team 7979

WardBenjamin, 2015-10-25:
My team has at least one dist. module with the same issues as described above by Kevin. He's right, inconsistant is the perfect way to describe it. Often, simply restarting the robot several times, or closing the app and turning the module off and back on fixes the issues related to non-detection of the hardware. 

The error is always something USB-related, usually that the software can't find a motor controller. We've never had any issues with power, though.

AlecHub, 2015-10-29:
Inconsistency is the nature of the race condition bug of the PDM [reported here](http://ftcforum.usfirst.org/showthread.php?4993-Update-on-USB-Disconnect-Issues-(unable-to-detect-USB-modules-during-scan)&p=18731&viewfull=1#post18731 "Race Condition Bug") over two weeks ago.

TheOutlaws, 2015-11-02:
And I am having troubles my team wants me to program the robot to go straight then turn and thiks is my first year doing android could I get a push in the right direction for this program

kokotc1, 2016-01-21:
We are experiencing this issue where we continually get the error message "Unable to open Modern Robotics USB DC Motor Controller..."  I've read other posts on the FTC forum as well that restarting usually resolves the issue, but we are not seeing this resolve.  Any advice for a rookie team that is supposed to complete in just 3 days who won't have time to send the module back for replacement?
Thanks!

lukethecoder64, 2016-01-21:
Try replacing the cable between the power distribution module and the
phone. Please note that the FTC forum is the best place to ask questions
like this, as that place gets lots of traffic.

>
>

-- 
Luke
     __
    /   /
   /   /
  /   /
 /   /______
/_________/

kokotc1, 2016-01-21:
We've tried three different cables, all with no impact...I have posted to
the FTC forum and also emailed Modern Robotics.  hopefully we'll find
something that works.

rwoodley, 2016-01-21:
This happens when the battery loses its charge.

TheOutlaws, 2016-01-21:
Well i had this problem it is your Modern power module

TheOutlaws, 2016-01-21:
When we returned our power module it stopped having problems also get a new
OTG cable

dumbomammoth, 2016-02-25:
We ordered a brand new power distribution module and one of the USB ports was bad and would not work upon the first try. I have now ordered the third power distribution module for our team. I will have to call and see what we can do to get this replaced. Had to pay 2nd day shipping because we are preparing for State Championship. Very frustrating for the kids.