ftctechnh/ftc_app #42: Color Sensoe

FTC0265, 2015-10-16:
We are unable to import the Color Sensor files even though they are in the directory. Can you fix this?

ftctechnh, 2015-10-16:
Hi FTC0265,

I am sorry that you are experiencing an issue.  I'm not sure if I understand your question.  Could you clarify what you are attempting to do?  Are you trying to write an op mode that uses the color sensor but are having problems?  Or have you written an op mode that uses a color sensor but you do not see them listed in the Op Mode selection on the Driver Station?

Also, I think you would get a better response and more people helping you if you were to post your question on the FTC Technology forum:

http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology

Would you mind reposting your question (with some additional detail) over on the FTC Technology forum?

Sincerely,

Tom

FTC0265, 2015-10-19:
We are trying to use a Modern Robotics Color Sensor with the newly updated github files. Whe we try to import the color sensor files using the import com.qualcomm.robotcore.hardware.ColorSensor; the ColorSensor part turns red and it says that it cannot recognize this symbol. Is the program just not finding the ColorSensor files in the directories?

Ace314159, 2015-10-19:
Go to `File`->`Invalidate Caches...` Then click on `Invalidate Caches and Restart`.

TheOutlaws, 2015-11-02:
And I am having troubles my team wants me to program the robot to go straight then turn and thiks is my first year doing android could I get a push in the right direction for this program

agent22922, 2015-11-03:
The colorsensor class is not available in the library, but there is examples of using it, javadoc etc.. Can someone point to a valid like for sdk download that has colorsensor class in it ?

Ace314159, 2015-11-03:
When you create an object using the colorsensor class, does the code build
properly?

agent22922, 2015-11-03:
Not able to build because it says that there are errors on the program

agent22922, 2015-11-03:
Never Mind when we did what @Ace314159 we didnt get an error

closed 2015-11-27 by gearsincorg