ftctechnh/ftc_app #45: Stopping OpMode does not remove stop button

calebsander, 2015-10-18:
I am not sure whether it happens every time, but it seems that usually pressing the stop button during the `loop()` stage of an OpMode successfully stops the OpMode but leaves the stop button visible on the screen. It is only when pressing the stop button again that it becomes the INIT button.
[image]

Sass-Meister, 2015-11-12:
Is the robot still doing things after you hit stop? Also, could you link the code that you're trying to run?

calebsander, 2015-11-12:
No, the OpMode's execution is stopped; it seems to be a simple UI bug. It happens regardless of what OpMode we use.

Sass-Meister, 2015-11-13:
If this still persists when using an opmode provided by First, than this is something they need to look at. Otherwise, this is a problem your code is creating.

calebsander, 2015-11-13:
I will test with a FIRST OpMode when I get a chance, but I don't see how our code could be the problem if the issue happens on the Driver Station app which we have not programmed at all.

calebsander, 2015-11-22:
It seems to be somewhat intermittent, and since we don't use any of the provided OpModes, I don't think I will really be able to reliably test whether it occurs more frequently with our OpModes.

gearsincorg, 2015-11-27:
I have seen several variations of this "incorrect GUI state transition" problem.
It does seem reasonable that it relates to opmode problems.  During early testing of the NavX sensor (when there were problems with memory and threads) this problem occurred pretty regularly.

calebsander, 2016-01-24:
Hmm. I haven't been able to reproduce lately, so I think I'll close this.

closed 2016-01-24 by calebsander