ftctechnh/ftc_app #47: Configuration files are stored separately from the selected file

calebsander, 2015-10-18:
The XML configuration files are stored in /sdcard/FIRST but the name of the selected one is not. This means that the configuration files persist when you update the app or build it from a new computer, but the selected file is not. For teams who plan to use multiple computers to build the app, it would be nice to save the name of the selected file so that they don't have to constantly re-activate it.

qwertychouskie, 2017-05-11:
I have never noticed this issue, probably been fixed.

calebsander, 2017-05-13:
Yeah, the issue still exists but the fact that the key used to sign the app is now included in the repository means that this poses less of a problem.

closed 2017-05-13 by calebsander