ftctechnh/ftc_app #49: Modern Robotics Color sensor throwing weird exceptions

Sass-Meister, 2015-10-20:
I'm trying to run the code colorsensordriver.java and after fixing some of the obvious issus like haveing the wrong maping and stuff I ran into an exception that didnt really give me much insite into the problem. "User code threw an uncaught exception: null" is the error it threw. If any more information is needed just ask. If anyone has had this happen to them and knows how to fix it, please let me know.

JacobAMason, 2015-10-20:
This could be an issue within some of your own code. Create a Gist (look at the top of this page) and paste in the opMode. (Alternatively, if you've already committed this code to your fork of the ftc_app repo, you can link us to the opMode in question.)

Sass-Meister, 2015-10-20:
This is the code i'm running to get a better understanding on the color sensor and the system this year

https://github.com/Crorem/ftc_app_5452/blob/master/FtcRobotController/src/main/java/com/qualcomm/ftcrobotcontroller/opmodes/demos/ColorSensorDriver.java

I'm still learning java so if this is some stupid mistake, let me know.

Sass-Meister, 2015-10-20:
Oh wait, I saw what I did wrong, commented it out and now it works.

closed 2015-10-22 by ftctechnh

TheOutlaws, 2015-11-02:
And I am having troubles my team wants me to program the robot to go straight then turn and thiks is my first year doing android could I get a push in the right direction for this program