ftctechnh/ftc_app #50: Driver Station App holds old gamepad values on startup.

gearsincorg, 2015-10-21:
This is an issue that effects robot/user safety.  
 When a particular series of events occur, the robot can start running expectantly, posing a safety risk.

The issue occurs when the user stops a running robot by hitting the phone's BACK or HOME button by mistake.

When this occurs, the driver station appears to simply stop sending control packets to the robot.  The stop() method is not called.  So, if the driver was driving firward quickly, the robot continues to drive forward even if the driver releases the joysticks.   After a short period, the robot detects the loss of communications and shuts down.

This run-on is somewhat serious but it's not completely unexpected.  HOWEVER the real safety issue occurs the next time the user goes to run the opmode.

If the driver re-selects the Driver Station app and runs it, (performing the usual opmode selection, init and play) the robot will start executing the last gamepad command it received, which may cause it to start driving forward at full speed (unexpected and dangerous).

How the user gains control of the robot will depend on how they terminated the program last time.  If they exited by pressing HOME, then the gamepads are still assigned and any joystick movement will regain control of the robot.  However, if they exited using BACK then they will need to re-assign the gamepad (using start A) before they can regain control.

FIX: It seems important that:
1) On seeing the app loosing focus, the stop() method should be called
2) On seeing an INIT Press, the Driver Station should clear out any defunct gampepad input.

ftctechnh, 2015-10-22:
Jonathan implemented a fix and it is being tested.

Jwierman94, 2015-10-22:
We have tested Jonathans fix and it will be included in the next release.

closed 2015-10-22 by Jwierman94

TheOutlaws, 2015-11-02:
And I am having troubles my team wants me to program the robot to go straight then turn and thiks is my first year doing android could I get a push in the right direction for this program