ftctechnh/ftc_app #52: "back" button on joystick not being mapped to gamepad.back

gearsincorg, 2015-10-27:
Even though the user icon glows green on the DS when you push the "back" button on the gamepad, it does not get reported through the gamepad object.

calebsander, 2015-10-28:
Can confirm. It also doesn't appear on the Robot Controller UI.

gearsincorg, 2015-11-07:
Incorrect ENUM being used.
Is using .BACK when it should be using .BUTTON_SELECT

calebsander, 2015-11-07:
Indeed. Should be a trivial fix. Relevant code:
```java
//In Gamepad (decompiled)
  public void update(KeyEvent event) {
    ...
    int var2 = event.getKeyCode();
    ...
    if (var2 == 4) {
      this.back = this.pressed(event);
    }
    ...
  }
```
```java
//In android.view.KeyEvent
public static final int KEYCODE_BACK            = 4;
...
public static final int KEYCODE_BUTTON_SELECT   = 109;
```

calebsander, 2016-01-07:
Decompiled code indicates that this has been fixed in the new release:
````
else if (var2 == 109) {
  this.back = this.pressed(event);
}
````

calebsander, 2016-01-09:
Can confirm that this has been fixed. Test program:
````java
public class BackButton extends OpMode {
	public void init() {}
	public void loop() {
		this.telemetry.addData("Back", this.gamepad1.back);
	}
}
````

closed 2016-01-11 by gearsincorg