ftctechnh/ftc_app #53: CDIM - unable to save configuration to file with more than one sensor configured.

AlecHub, 2015-10-28:
When configuring the CDIM with more than one sensor (say COLOR_SENSOR and TOUCH_SENSOR), I am unable to save the configuration due to the following error:

[image]

Saves fine when only one sensor  is configured.  The log file simply echos the same error.

AlecHub, 2015-10-29:
I should have known that the sensors need to be named.  Naming the sensors resolves the issue.

Perhaps this ticket can be a feature request for the UI to prevent configuring a sensor with no name.

gearsincorg, 2015-11-27:
Or... it would be helpful to provide some more information as to which type of device has the blank names...

AlecHub, 2016-02-11:
I believe I mentioned it was a COLOR_SENSOR and TOUCH_SENSOR.  

BTW, how did you get to be a "collaborator", and why can't non-collaborators assign labels to their own tickets?

gearsincorg, 2016-02-13:
AlexHub.  You miss-read my response.  I was suggesting that the Error message should provide more information regarding which devices had blank names, not you ;)

Error messages that don't give you enough information to correct the problem are not very useful.
Sorry for the confusion.

Phil.

AlecHub, 2016-02-16:
Hi Phil... Yes, I misread your comment, and yes more info to correct the problem is needed.  Thanks, Alec.

gearsincorg, 2016-09-04:
Closed, as this issue relates to previous season code.

closed 2016-09-04 by gearsincorg