ftctechnh/ftc_app #57: Robot controller app is throwing Android exceptions

ZachOhara, 2015-10-31:
The robot controller app is consistently throwing exceptions during robot initialization. The exact error seems to be a problem in the way that the app is interfacing with the Android SDK. Our team is running the app on a ZTE Speed with Android version 4.4.4, and  version 1.2 of the FTC robot controller. The full stack trace of the exception is shown below:

```
Activity com.qualcomm.ftcrobotcontroller.FtcRobotControllerActivity has leaked IntentReceiver com.qualcomm.analytics.Analytics@420f8aa8 that was originally registered here. Are you missing a call to unregisterReceiver()?
android.app.IntentReceiverLeaked: Activity com.qualcomm.ftcrobotcontroller.FtcRobotControllerActivity has leaked IntentReceiver com.qualcomm.analytics.Analytics@420f8aa8 that was originally registered here. Are you missing a call to unregisterReceiver()?
	at android.app.LoadedApk$ReceiverDispatcher.<init>(LoadedApk.java:796)
	at android.app.LoadedApk.getReceiverDispatcher(LoadedApk.java:597)
	at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1520)
	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1500)
	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1494)
	at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:467)
	at com.qualcomm.analytics.Analytics.register(SourceFile:209)
	at com.qualcomm.analytics.Analytics.<init>(SourceFile:235)
	at com.qualcomm.modernrobotics.ModernRoboticsUsbUtil.init(SourceFile:85)
	at com.qualcomm.ftccommon.FtcEventLoop.init(SourceFile:89)
	at com.qualcomm.robotcore.eventloop.EventLoopManager.a(SourceFile:496)
	at com.qualcomm.robotcore.eventloop.EventLoopManager.setEventLoop(SourceFile:417)
	at com.qualcomm.robotcore.eventloop.EventLoopManager.start(SourceFile:369)
	at com.qualcomm.robotcore.robot.Robot.start(SourceFile:58)
	at com.qualcomm.ftccommon.FtcRobotControllerService$b.run(SourceFile:160)
	at java.lang.Thread.run(Thread.java:841)
```

The full log file that this excerpt was taken from is available [here](https://gist.github.com/PranavMathur/52923dde67a212ac0bc0). The included exception occurs on line 7746 of the file.

Our team is unable to test anything at the moment, because the app will not survive past the initialization routine. We will continue to look into additional causes of the error, but in the event that FTC's code is the fundamental cause of the error, teams will be almost completely powerless to fix it.

dmssargent, 2015-11-01:
Did you remove try to remove the FTC tracking files? Also, can you give us a truncated logcat of just one app run?

Sorry, but what are trying to do in that log, newInstance works by calling the parameterless constructor in the class, and you sure do seem to be having an insane amount of GC'ing, don't forget that Android has an extremely limited heap compared to a normal computer's available heap.

gearsincorg, 2016-09-04:
Closed, as this issue relates to previous season code.

closed 2016-09-04 by gearsincorg