ftctechnh/ftc_app #59: Auto Drive

TheOutlaws, 2015-11-02:
And I am having troubles my team wants me to program the robot to go straight then turn and this is my first year doing android could I get a push in the right direction for this program it would be very helpfull

magneticflux-, 2015-11-02:
Well, you need to first name the motors in the robot controller. Once they are named, access them by using hardwareMap.dcMotor.get("motor_name"), with the name surrounded in quotes. Then, you can use the setPower method in the DcMotor class to control the motor you retrieved. Your OpMode should probably extend LinearOpMode so you can wait for time. Now, you can set both motors powers' and wait for 1000ms to make the robot advance.

dlevyaaas, 2015-11-02:
Probably better to post questions here:  http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology

TheOutlaws, 2015-11-02:
thx here is what I have so far 

code: 
/* Copyright (c) 2014 Qualcomm Technologies Inc

All rights reserved.

Redistribution and use in source and binary forms, with or without modification,
are permitted (subject to the limitations in the disclaimer below) provided that
the following conditions are met:

Redistributions of source code must retain the above copyright notice, this list
of conditions and the following disclaimer.

Redistributions in binary form must reproduce the above copyright notice, this
list of conditions and the following disclaimer in the documentation and/or
other materials provided with the distribution.

Neither the name of Qualcomm Technologies Inc nor the names of its contributors
may be used to endorse or promote products derived from this software without
specific prior written permission.

NO EXPRESS OR IMPLIED LICENSES TO ANY PARTY'S PATENT RIGHTS ARE GRANTED BY THIS
LICENSE. THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
"AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO,
THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE
FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE. */

package com.qualcomm.ftcrobotcontroller.opmodes;

import com.qualcomm.robotcore.eventloop.opmode.OpMode;
import com.qualcomm.robotcore.hardware.DcMotor;
import com.qualcomm.robotcore.util.Range;

/**
 * TeleOp Mode
 * <p>
 * Enables control of the robot via the gamepad
 */
public class AidenTestv2v2 extends OpMode {

	/*
	 * Note: the configuration of the servos is such that
	 * as the arm servo approaches 0, the arm position moves up (away from the floor).
	 * Also, as the claw servo approaches 0, the claw opens up (drops the game element).
	 */
    // TETRIX VALUES.

    // position of the arm servo

    // amount to change the arm servo position.

    // position of the claw servo

    // amount to change the claw servo position by

    DcMotor motorRight;
    DcMotor motorLeft;

    /**
     * Constructor
     */
    public AidenTestv2v2() {

    }

    /*
     * Code to run when the op mode is first enabled goes here
     *
     * @see com.qualcomm.robotcore.eventloop.opmode.OpMode#start()
     */
    @Override
    public void init() {

		/*
		 * Use the hardwareMap to get the dc motors and servos by name. Note
		 * that the names of the devices must match the names used when you
		 * configured your robot and created the configuration file.
		 */

		/*
		 * For the demo Tetrix K9 bot we assume the following,
		 *   There are two motors "motor_1" and "motor_2"
		 *   "motor_1" is on the right side of the bot.
		 *   "motor_2" is on the left side of the bot and reversed.
		 *
		 * We also assume that there are two servos "servo_1" and "servo_6"
		 *    "servo_1" controls the arm joint of the manipulator.
		 *    "servo_6" controls the claw joint of the manipulator.
		 */
        motorRight = hardwareMap.dcMotor.get("motor_2");
        motorLeft = hardwareMap.dcMotor.get("motor_1");

        motorLeft.setDirection(DcMotor.Direction.REVERSE);

        // assign the starting position of the wrist and claw
    }

    /*
     * This method will be called repeatedly in a loop
     *
     * @see com.qualcomm.robotcore.eventloop.opmode.OpMode#run()
     */

    @Override
    public void loop() {

    }

}

magneticflux-, 2015-11-02:
@TheOutlaws It's rather rude to post the same request on many issues. I would advise you to first ask your teammates for aid. It that fails, the forums are the correct location for help.

TheOutlaws, 2015-11-03:
k sry but why that now I updated all of my and default opmodes are showed up as obslete

KingofGamesYami, 2015-11-03:
Um, could you please define a few things for me?

1.  How do you know your opmodes are obsolete?  Like, how do they show up as obsolete?  I don't think that's the word you're looking for...
2.  How did you update?  Git, downloading the SDK, etc.  If you downloaded the SDK, did you move your opmodes into it?
3.  Did you check to see if you are using manager.register to register your opmodes?
4.  How old are you?  You sound like a 12-year-old texting (no offense).

TheOutlaws, 2015-11-03:
I fixed it the Api 16 needed updated and second of all I am 12 jerk and I also got the program to work

WardBenjamin, 2015-11-03:
@TheOutlaws Okay, you need to chill. Not only are you the one asking for help on an issue that no one else has, you were very rude yourself by posting your question on every github issue. You are asking unintelligable questions, and @KingofGamesYami was trying to help you; You should NOT have insulted him. From a FIRST view, you need to learn some gracious professionalism.

closed 2015-11-04 by TheOutlaws