ftctechnh/ftc_app #68: No Type HiTechnicNxtGyroSensor for HardwareMap

aeleos, 2015-11-06:
There is no type HiTechnicNxtGyroSensor for HardwareMap. Using gyroSensor instead when the object is created like ```HiTechnicNxtGyroSensor GyroSens``` returns the error  
```Error:(81, 43) error: incompatible types
required: HiTechnicNxtGyroSensor
found:    GyroSensor```
and using the type HiTechnicNxtGyroSensor returns the error 
```Error:(81, 28) error: cannot find symbol variable HiTechnicNxtGyroSensor```

I am fairly sure I explored all the possibilities and its actually a bug, but I am not sure as I can't see the source code for the hardwaremap function.

calebsander, 2015-11-06:
You need to cast the object if you want it as a `HiTechnicNxtGyroSensor`, although there is no reason to since that class doesn't have any methods other than those in `GyroSensor`.

aeleos, 2015-11-07:
Unless I am understanding it wrong what you should be doing is something like
```HiTechnicNxtGyroSensor gyro;```
```gyro = hardwareMap.HiTechnicNxtGyroSensor.get("gyro_1");```
which returns the error ```Error:(81, 28) error: cannot find symbol variable HiTechnicNxtGyroSensor```

calebsander, 2015-11-07:
Really what you should be doing is GyroSensor sensor = this.hardwareMap.gyroSensor.get("sensorName");

aeleos, 2015-11-07:
And so that will work even if I am using the NXT Gyro Sensor?

calebsander, 2015-11-07:
You will get an instance of GyroSensor. If your configuration file specifies that the name you provided is for a HiTechnic gyro, then that object will be of type HiTechnicNxtGyroSensor, even though it is being store in a variable of type GyroSensor. When you call getRotation( on it), it will return the result of calling getRotation() on any object of type HiTechnicNxtGyroSensor.

calebsander, 2015-11-08:
Can this be closed?

closed 2015-11-27 by gearsincorg