ftctechnh/ftc_app #71: Repo is a full blown 200MB

Schemetrical, 2015-11-07:
The repo is bloated with useless things such as 100MB worth of git history and this 70MB pdf full of images. Those should be linked, not included in the repo. This is terrible practice and a waste of bandwidth :(

calebsander, 2015-11-07:
I agree that it is unfortunate that there is so much in this repository that is not for the app itself or the Javadoc, however I don't think that the problem can be rectified without making a new repo, since the history cannot be undone. For future seasons, I think it would be best to put the APK files and the instruction PDFs on a separate hosting service. In addition, this could probably be helped significantly by storing the source code for the SDK instead of the compiled files. Whenever a change is made to any code in one of the AAR files, it necessitates saving a new version of the entire file.

magneticflux-, 2015-11-07:
The bandwidth pain is slightly rectified by Git's fundamental changes-only property, but when every commit @ftctechnh makes forces a re-upload of _every file_, it becomes an issue. Perhaps uploading the FTC libraries to Maven Central would be better? Then each team would only have to add a "compile: 'example:project:10.0.1'" statement to their build.gradle.

cchan, 2015-11-07:
The best solution would be to restructure such that you only need to download the package once, and then git init a repository in a subfolder. Thus, Git will only track your own code, and this ftc_app repository will only track the actual app code.

It would also help a lot if Qualcomm released the source code, but I'm sure there's some copyright problems. FTC is kinda supposed to be an exemplar of gracious professionalism, though, and I would think that Open Source is the highest incarnation of GP. Gotta compromise, though.

Schemetrical, 2015-11-08:
Also [this is pretty cool](http://stackoverflow.com/questions/2100907/how-to-remove-delete-a-large-file-from-commit-history-in-git-repository).

`$ java -jar bfg.jar --strip-blobs-bigger-than 50M my-repo.git`

strugee, 2015-11-13:
@Schemetrical you don't need to use some random Java tool for this; Git can do it by itself via `git filter-branch`. That being said, don't do that. If you do, you will break the clones (and forks) of every single person who has ever made a copy of this repository.

In the future, though, [Git LFS][1] would be a great solution to this.

 [1]: https://git-lfs.github.com/

Schemetrical, 2015-11-13:
@strugee I know, obviously it wouldn't work in this case because its already broke but whoever committed that massive pdf must have thought "oh its taking forever... fuck it"... pretty disappointing but you can't do anything about it :/

clwillingham, 2016-09-03:
So this is not really a solution, but it is an option. I created a separate copy of this repository without the history or documentation: https://github.com/clwillingham/ftc_app_only
current repository size is 2.7mb
Read the readme.md in the repository for more information on keeping it up to date.

calebsander, 2016-09-04:
You can get around the large repository size on the initial download by just downloading the zip file of the current contents instead of cloning the repository. If you want to fork the repository, you can get rid of the history by instead unzipping the downloaded zip file into a new repository. I think FIRST should also put the beta apk files onto the Play Store and remove them from the repository. The Robot Controller apk file is especially not needed since most teams cloning this repository are going to be building it themselves. To keep the commit size down in the future now that the source code has been released, it would also make sense to supply Java files for the SDK rather than the compiled archives so that the files can be diffed.

rgatkinson, 2016-09-04:
We're working on reducing the size of the repo. There are a surprising number of issues to consider, as the FTC community is a diverse one, and the workflow that works optimally for one team isn't necessarily the best one for another. For example, there is significant value for many teams in having a one-stop-shopping cloneable/forkable/downloadable package. Conversely, for most teams, being able to completely-faithfully revert, say, to the exact release state as of Christmas 2015 is (now) of little value.

Our current favorite approach is to remove the largest obsolete files from the GIT history (with the help of the BFG tool mentioned by @Schemetrical). Removing just the obsolete PDFs, for example, results in a 58% savings in cloning size (100.12MB instead of 239.83MB). We hope to be able to reduce this even further.

Schemetrical, 2016-09-04:
@rgatkinson thanks so much :)

strugee, 2016-09-04:
@rgatkinson sounds like a plan, though I'd like to once again point out that `git filter-branch` can also do what you're trying to accomplish. If I were FTC I'd use the built-in Git tools when possible - it just reduces the chances of something going wrong.

cchan, 2016-09-04:
BFG is a well-known and very widely used subset of `git filter-branch` that does the desired task of scrubbing a Git history. It's highly reliable, and if anything goes wrong you can just `git reset` to the old `HEAD` hash anyway. `git filter-branch` can do much more powerful things besides that, but that makes it much slower. Insanely slower. Legitimately hundreds of times slower.

strugee, 2016-09-04:
@cchan OK. I haven't heard of BFG even once before this thread, but I'll take your word for it.

rgatkinson, 2016-09-04:
Per [GitHub](https://help.github.com/articles/remove-sensitive-data/), "The BFG Repo-Cleaner is a faster, simpler alternative to git filter-branch for removing unwanted data."

cmacfarl, 2016-09-04:
BFG = acronym(reverse("Git Filter Branch"));

This is a relatively good introduction.

https://www.theguardian.com/info/developer-blog/2013/apr/29/rewrite-git-history-with-the-bfg

richiksc, 2016-09-25:
I think the documentation can safely be moved to GitHub Pages along with the media. For offline access, there should be downloadable resources from the FIRST website.

Schemetrical, 2017-02-25:
Well #228 broke the repo so it got reverted, but the issue still remains. I'm not sure how to proceed other than to branch off and prune.

Schemetrical, 2017-02-25:
for now use `git clone https://github.com/ftctechnh/ftc_app.git --depth 1`

richiksc, 2017-02-25:
@Schemetrical - A good idea is to fork the repo, clone, create an orphan GH pages branch and delete its contents by rmrfing the whole thing. Then, migrate the documentation and media and create a landing page. I'm on it.

cmacfarl, 2017-02-25:
If all you are after is the latest master, and you don't care about the alpha or beta branches then @Schemetrical has the right answer.  That trims everything down to about 62M.  FIRST is still looking at a couple of different options to reduce the size of the repository, but nothing is likely to happen until after worlds.

JacobAMason, 2017-06-14:
@cmacfarl Any word on a repo redo? (Don't want repo urls to change overnight and break the Net installer for everyone.)

cmacfarl, 2017-06-15:
The option that was looked at was git LFS.  Unfortunately, there's no native support for git LFS in the git clients or github itself.  So if you clone a repo from github that has git LFS stored objects, the clone doesn't automatically fetch the LFS objects.   You must also install a git LFS extension.  This was deemed an unreasonable burden, for solid reasons, and was hence rejected in favor of publicizing the --depth option for those that want to restrict the size of the repo download.

JacobAMason, 2017-06-15:
Okay. That seems like a reasonable conclusion for this issue, if you want to close it.

Schemetrical, 2017-06-15:
Better it break now then during season IMO

Schemetrical, 2017-06-15:
My bad, the message sent late. I guess the depth option works, but imo I’d rather have the LFS files not clone because I wouldn’t need them anyways

closed 2017-06-15 by Schemetrical