ftctechnh/ftc_app #75: PID control in the android system

Sass-Meister, 2015-11-11:
When we were using robotc last year, all we had to do to activate PID control. Is there a system like this for the new system? I skimmed through the documentation and there doesn't see to be any support for it except things like motor.setEncoderTarget();.PID control was super useful and new teams may not even realize they need to put a system in place to smooth out the motor.

KyleDavidE, 2015-11-11:
`motor.setMode(DcMotorController.RunMode.RUN_USING_ENCODERS);` should do the trick.

Note that in past versions it has been `setChannelMode` but that naming has been deprecated.

- 8391 W-Prime robotics.

Sass-Meister, 2015-11-11:
Ok thanks

calebsander, 2015-11-21:
Can this be closed?

closed 2015-11-27 by gearsincorg