ftctechnh/ftc_app #79: OpMode transfer between Robot Controller and Driver Station not sending enough data

danielgrimshaw, 2015-11-14:
After many hours of testing for problems relating to https://github.com/ftctechnh/ftc_app/issues/70, we have found the error to be in a limit of data transfer size when the Robot Controller transfers OpMode's to the Driver Station. Currently, it is limited to 256 bytes of data, which is not enough for use of the [Thunderbots Robot Controller](https://github.com/thunderbots5604/Thunderbots-Robot-Controller) with a large number of OpModes. Is it possible to set the limit to something impossibly high, say around 5KiB, or just completely remove the limit?

calebsander, 2016-01-07:
Can you test in the beta release? It says "code modified to allow for a long list of op mode names."

calebsander, 2016-01-09:
Can confirm that it is fixed in the new release:
```java
manager.register("Extremely1 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely2 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely3 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely4 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely5 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely6 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely7 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely8 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely9 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely0 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
manager.register("Extremely11 Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name Extremely Long OpMode Name", new BlockPartyDrive());
````

closed 2016-09-04 by gearsincorg