ftctechnh/ftc_app #82: Restart Robot not working from the Driver Station

levydev, 2015-11-16:
It is necessary to perform a "Restart Robot" operation after downloading to the Robot Controller.  Unfortunately this only works from the Robot Controller phone.  This makes it hard for wireless downloading or even for wired downloads in cases where it is difficult to reach the menu buttons on the screen.

There is an equivalent ""Restart Robot"  menu option on the Driver Station phone that does not seem to do anything.

Can this be fixed?

KingofGamesYami, 2015-11-18:
I have successfully used the restart robot command from the Drive Station.  However, it's hard to tell when the Robot Controller has restarted - you have to wait and hope.

rgatkinson, 2016-09-04:
@levydev I believe this is now fixed.

closed 2016-09-04 by rgatkinson