ftctechnh/ftc_app #90: DriverStation & RobotController Sanity Checks

AlecHub, 2015-11-23:
Feature request for DriverStation (DS) & RobotController (RC) "sanity checks":
* DS & RC check for FTC legal WiFi-Direct names:
 * RC doesn't create WiFi-Direct group unless WiFi-Direct name of RC is FTC legal, and
    * no newline characters
 * DS doesn't attempt to pair with RC unless WiFi-Direct name of DS is FTC legal, and
    * no newline characters
    * team number portion of RC name matches team number portion of DS name
* DS checks version of RC and disconnects if DS version is incompatible with RC version
* Any other useful check that the DS and RC can make

rwoodley, 2015-11-23:
I don't think we should introduce more ways for WiFi pairing to fail silently. Sanity check failures could be logged, but should not result in WiFi errors.

AlecHub, 2015-11-24:
If there is a newline character in the WiFi-Direct name, then it crashes other teams' phones. In this case, the RC should not create a WiFi Direct group, and the DS should not attempt to pair with any RCs, if either of their respective names has a newline character.

Agree with @rwoodley that WiFi pairing could be logged instead of fail for the other cases if by "logged" you mean a big "red alert" indication in the GUI; not just some message buried in the logcat.

strugee, 2015-11-25:
@rwoodley so don't fail silently. Fail loudly, and with lots of error dialogs.

magneticflux-, 2015-11-25:
@strugee The problem is, it shouldn't bug people when other people's malformed names try to connect. It should hide the error on the receiving end, and show it on the sending end. No problem finding the error if both phones are from one team, but other people can't be malicious with the system.

gearsincorg, 2016-09-04:
Closed, as this issue relates to previous season code.
Current season includes built-in inspection tool & optional name filtering.

closed 2016-09-04 by gearsincorg