ftctechnh/ftc_app #92: Extra servo is created

calebsander, 2015-11-26:
This is my configuration file:
````
<?xml version='1.0' encoding='UTF-8' standalone='yes' ?>
<Robot>
    <MotorController name="Accessories" serialNumber="AL00VD65">
        <Motor name="bucket" port="1" />
        <Motor name="arm" port="2" />
    </MotorController>
    <DeviceInterfaceModule name="DIM" serialNumber="AL00VDDK">
        <I2cDevice name="bno055" port="0" />
    </DeviceInterfaceModule>
    <MotorController name="Left" serialNumber="AL00VESE">
        <Motor name="F" port="1" />
        <Motor name="G" port="2" />
    </MotorController>
    <LegacyModuleController name="Legacy Module" serialNumber="AL00VBU7">
        <ServoController name="Servos" port="0">
            <Servo name="roller" port="1" />
            <Servo name="climber" port="2" />
        </ServoController>
    </LegacyModuleController>
    <MotorController name="Right" serialNumber="AL00VFH4">
        <Motor name="D" port="1" />
        <Motor name="E" port="2" />
    </MotorController>
</Robot>
````
There is a servo controller on a Legacy Module with servos attached on the first two channels. However, the `HardwareMap` indicates that there are servos on ports 1, 2, and 6. When I iterate over all the servos in `hardwareMap.servo` and print out the connection info, I get the following:
````
USB AL00VBU7; port 0; port 6
USB AL00VBU7; port 0; port 1
USB AL00VBU7; port 0; port 2
````
The ghost servo also appears in the app's logcat:
````
========= Device Information ===================================================
Type                                          Name                           Connection
Modern Robotics USB DC Motor Controller       Right                          USB AL00VFH4
Modern Robotics USB DC Motor Controller       Accessories                    USB AL00VD65
Modern Robotics USB DC Motor Controller       Left                           USB AL00VESE
Analytics detected NetworkInfo.State.CONNECTED
DC Motor                                      D                              USB AL00VFH4; port 1
DC Motor                                      E                              USB AL00VFH4; port 2
DC Motor                                      F                              USB AL00VESE; port 1
DC Motor                                      G                              USB AL00VESE; port 2
DC Motor                                      arm                            USB AL00VD65; port 2
DC Motor                                      bucket                         USB AL00VD65; port 1
NXT Servo Controller                          Servos                         USB AL00VBU7; port 0
Servo                                         NO DEVICE ATTACHED             USB AL00VBU7; port 0; port 6
Servo                                         roller                         USB AL00VBU7; port 0; port 1
Servo                                         climber                        USB AL00VBU7; port 0; port 2
Modern Robotics USB Legacy Module             Legacy Module                  USB AL00VBU7
Modern Robotics USB Device Interface Module   DIM                            USB AL00VDDK
I2cDevice                                     bno055                         USB AL00VDDK; port 0
Modern Robotics USB DC Motor Controller       Right                          USB AL00VFH4
Modern Robotics USB DC Motor Controller       Accessories                    USB AL00VD65
Modern Robotics USB DC Motor Controller       Left                           USB AL00VESE
````
For teams who are only getting servos by name, this probably isn't such a problem, but it becomes apparent when trying to target all the servos regardless of the configuration file.

calebsander, 2016-01-09:
Still an issue in 16.01.04. Seems to always happen on the last port of a servo controller attached to a Legacy Module.

Team10824Mentor2, 2016-01-15:
Our FTC team is struggling to add an additional servo.  Where do we find the configuration file that you have outlined above.  Also, can this file be revised such that we call out that the newly added servo uses port 6 on the core servo controller

CarteriTrellis, 2016-01-15:
@Team10824Mentor2 Let's take this offline. Go to my profile page; my email address is my first name @ the domain listed on my profile.

rgatkinson, 2016-09-04:
@calebsander Is this still an issue that you see?

cmacfarl, 2016-11-14:
@calebsander Please feel free to reopen if this is still an issue.

closed 2016-11-14 by cmacfarl