ftctechnh/ftc_app #94: Documentation needs to be improved.

cn04, 2015-11-29:
There is a saying popular among computer programmers that goes something like this:
<i>Don't name the class "AccelerationSensor" and make the description "Acceleration Sensor".</i>  
There's also a saying that goes like:
<i>Don't name the class "PWMOutputController" and make the description "Interface for working with PWM Input Controllers".</i>  
There's also a saying that goes like:
<i>Don't name the class "CompassSensor" and make the description "Compass Sensor".</i>  
There's also a saying that goes like:
<i>Don't name the class "IrSeekerSensor" and make the description "IR Seeker Sensor".</i>  
There's also a saying that goes like:
<i>Don't name the class "LightSensor" and make the description "Light Sensor".</i>  
Countless more occurrences of this type of "documentation" can be found in this API.  
A description is supposed to describe something. It isn't supposed to just be the same thing as the name of the class. The same applies to methods. Why would you create a method called getVoltage() with the description "Get the current voltage"? In what form? Volts? Millivolts? Will it ever return a negative voltage? It's a bit hard to get anything done when nobody knows what this API is intended to be used for. Is the "LightSensor" class for NXT light sensors connected to the Legacy Module? For Modern Robotics light sensors? For both? For EV3 light sensors? Is the "IrSeekerSensor" class for the HiTechnic IRSeeker sensor or for the Modern Robotics one? What the heck is a "PWMOutputController"? Is the "GyroSensor" the new EV3 gyro sensor? Is it an IMU connected to the Device Interface using I2C? Is it some other form of gyro sensor?  

Nobody knows except the programmers who created the API. I understand that this is a new platform, and this is the first year of its existence, but I have yet to obtain the skill of being able to read the minds of other programmers.

KingofGamesYami, 2015-11-30:
I feel the same - documentation is very poor.  I want a lot more of "When you do x, this happens" rather than "Is the motor busy?".  How do I know what constitutes the motor being busy?  At first glance, this might look like it's checking whether or not the motor is trying to move - which is useless, since we don't need a function to determine that.  So what does it actually do?  I don't know...  I get 30 minutes before each competition to program our robot.

cn04, 2015-11-30:
Yes. Also, I noticed that it's unclear which type of hardware many of the classes in the robotcore 'hardware' package are referring to. Is the "LightSensor" an NXT or Modern Robotics light sensor? What kind of gyro sensor is the "GyroSensor"? HiTechnic? LEGO? and so on.

KingofGamesYami, 2015-11-30:
Well, I can tell you right now that those classes cover multiple sensors.  The only sensor I know of that isn't supported is the LEGO Color Sensor (note: HiTechnic Color Sensor is supported).  So using the LightSensor class will work on both LEGO and Modern Robotics light sensor AFAIK.

Bendito999, 2016-01-13:
No more +1's please! It does not accomplish anything, and fills up everyone's inbox. Thank you.

strugee, 2016-01-13:
@ftctechnh please consider locking this issue to prevent further +1s.