ftctechnh/ftc_app #97: It is possible to register any Class as an OpMode

calebsander, 2015-12-03:
Due to the use of reflection to store OpModes, trying to run a non-OpMode class as an OpMode throws some cryptic errors at runtime. It seems like this could be pretty easily solved by changing the method signature of `OpModeManager.register()` from `public void register(String name, Class opMode)` to `public void register(String name, Class<? extends OpMode> opMode)`.

strugee, 2015-12-07:
@calebsander if you have the code to solve this (anything, really), you should consider sending a Pull Request!

calebsander, 2015-12-07:
The issue is that the code is contained in one of the precompiled libraries, so it isn't really possible for the general public to submit pull requests until they decide to open-source the code.

calebsander, 2016-01-07:
Still an issue in 2016.01.04

calebsander, 2016-02-11:
Still an issue in yesterday's release

calebsander, 2016-04-06:
I would recommend removing the class-based register method in favor of the instance-based method because it makes explicit when the constructor is called.

magneticflux-, 2016-04-06:
Perhaps something similar to how the ```Runnable``` or ```Function<E, F>``` interfaces are used. On the other hand, that would complicate providing instances of them. A no-args constructor useable via reflection might be the best option, barring some sort of Java 8-esque ```Supplier<? extends OpMode>``` system.

calebsander, 2016-08-20:
Fixed in 2.0.

closed 2016-08-20 by calebsander