ftctechnh/ftc_app #99: hardwarefactory missing in latest release, but I need dat gyro.

RoboticColonelsComp02, 2015-12-03:
Hello Github!

I have recently been trying to use a gyro with our autonmous code (which should be easy) but the GyroSensor class was empty. I did some googleing and found that this was "fixed" in the latest release. I immediately downloaded the zip and very *very* carefully combined my working directory with the update. I know my way around android projects, so this was pretty straight-forward. After updating, however, FtcRobotControllerActivity.java (which i pulled from the update) started throwing errors about the library `com.qualcomm.hardware.HardwareFactory` not existing. This is a problem. The latest release should come with working files. Does anyone know what AAR files contains this, or know of another way to fix this?

magneticflux-, 2015-12-03:
@RoboticColonelsComp02 It's very likely something to do with how you updated your project. Instead of merging ftc_app into your old app, merge your team-specific code (it should be all located in one place and easily moveable) into an updated ftc_app.

RoboticColonelsComp02, 2015-12-03:
@magneticflux that is what I did. Like I said, I was very careful, and I know my way around an android project.
as a note:
when I was looking through the AARs, i found that `hardware-release.aar` __*has the class*__ `com.qualcomm.hardware.HardwareFactory`!

RoboticColonelsComp02, 2015-12-03:
I figured it out. I checked the module dependencies of the project, and `Hardware-release.aar` was missing. Hmm, time to use __File > New Module > Import .JAR/.AAR Package__

closed 2015-12-03 by RoboticColonelsComp02