ftctechnh/ftc_app #102: The function getCurrentPosition() always returns zero with the legacy motor controller.

FTC6499, 2015-12-05:
It appears that the function <code>getCurrentPosition()</code> does not ever update inspite of the fact that motor moves and in spite of the fact that rotations are being recongized by the rotatry encoder. PID is working so the motor controller must be registering the rotations in some way. I know that PID is working because the motor ramps up it's power after moving.

Here are the related global variables.
``` java
    DcMotor[] arm;
    DcMotorController armzeroctl;
    int armzeroid;
    boolean whichArm=true;
    boolean quearmize;
    double speedArm=6.;
    double armzeropos;
    int lastarmzeropos;
    boolean armzerobusy;
```
Some of these variables are initalized in <code>init()</code> as so. I based my code off of the <code>K9TeleOp</code> class.
```java
       arm=new DcMotor[2];

        arm[0] = hardwareMap.dcMotor.get("arm0");
        arm[0].setMode(DcMotorController.RunMode.RUN_USING_ENCODERS );

        armzeroctl=arm[0].getController();
        armzeroid=arm[0].getPortNumber();
```
In <code>loop()</code> I have the following code
```java
           if(armzeroctl.getMotorControllerDeviceMode()==DcMotorController.DeviceMode.WRITE_ONLY) {
                if((!armzerobusy)&&quearm) {
                    int armposround=(int)armzeropos;
                    arm[0].setPower(.01);
                    arm[0].setTargetPosition((int) armzeropos);
                    quearm=false;
                    lastarmzeropos=(int)armzeropos;
                }
                arm[1].setPower(scaleInput(gamepad2.right_stick_y));
                armzeroctl.setMotorControllerDeviceMode(DcMotorController.DeviceMode.READ_ONLY);
            }else if(armzeroctl.getMotorControllerDeviceMode()==DcMotorController.DeviceMode.READ_ONLY){
                armzeropos -= (scaleInput(gamepad2.left_stick_y) / speedArm * 32.);
                if(((int)armzeropos)!=lastarmzeropos){
                    quearm=true;
                }
                armzerobusy=arm[0].isBusy();
                telemetry.addData("Arm", String.format("Position %d Target copy: %d Target: %d busy: %b", arm[0].getCurrentPosition(),arm[0].getTargetPosition(), (int) armzeropos,armzerobusy));
                armzeroctl.setMotorControllerDeviceMode(DcMotorController.DeviceMode.WRITE_ONLY);

            }
```
Upon viewing the telmetry data I noticed that <code>getTargetPosition()</code> is always zero and most likely because of this the motor never stops. I used the Netbeans decompiler and I am baffled by this issue because of the following:
```java
    public int getMotorTargetPosition(int motor) {
        this.a(motor);
        this.b();
        byte[] var2 = new byte[4];

        try {
            this.c.lock();
            System.arraycopy(this.b, OFFSET_MAP_MOTOR_TARGET_ENCODER_VALUE[motor], var2, 0, var2.length);
        } finally {
            this.c.unlock();
        }

        return TypeConversion.byteArrayToInt(var2);
    }

    public int getMotorCurrentPosition(int motor) {
        this.a(motor);
        this.b();
        byte[] var2 = new byte[4];

        try {
            this.c.lock();
            System.arraycopy(this.b, OFFSET_MAP_MOTOR_CURRENT_ENCODER_VALUE[motor], var2, 0, var2.length);
        } finally {
            this.c.unlock();
        }

        return TypeConversion.byteArrayToInt(var2);
    }
```
This is from <code>HiTechnicNxtDcMotorController.class</code> and we can see that <code>getMotorTargetPosition()</code> is using reading from the array which is called <code>b</code>. Values returned by <code>getMotorTargetPosition()</code> are correct. However <code>getMotorCurrentPosition()</code> always returns zero. At this point I do not know how to percede futher. What could be causing this issue?

gearsincorg, 2015-12-05:
It's very possible that your encoders are not connected correctly. And therefore not generating pulses to be counted.  The code sample shows reading  two sets of numbers from the motor controller hardware registers.  This looks perfectly OK.

You say that "I know that PID is working because the motor ramps up it's power after moving."

But this in not conclusive.   If you are requesting a constant slow speed, and the encoders are not connected properly, then the motor controller will start with low power, and then continue to ramp up the power in an attempt to generate encoder counts.  Eventually it will max out the power.  To me, this an indication of a disconnected encoder.

Please verify that the encoder is plugged into the correct motor channel (1 or 2) and that the encoder cable has been plugged in the correct way around (yellow wire near the outside edge of the motor controller).

I true indication that the encoders are being read correctly is that you can run the motors slow for several seconds and run them fast for several seconds.

FTC6499, 2015-12-09:
I just verified that it is not a hardware issue. Instead it is a software bug (either on my part or in compiled FTC provided code).

It seems to be a combination of two problems
* Not reseting the encoders
* Not waiting for pending writes before switching to READ_ONLY mode.

The reason I know that this is a software bug is because I wrote a test program in which I can see getCurrentPosition() correctly updating by adding a delay before switching to READ_ONLY mode. I verified that it works
The initalization is the same except I start with RESET_ENCODERS and later switch to RUN_TO_POSITION. I added a global variable called loopcnt which is of type int.
Here is my code for loop.
```java
            if(loopcnt==24) {
                arm[0].setMode(DcMotorController.RunMode.RUN_TO_POSITION  );
            }else if(loopcnt==64) {
                telemetry.addData("Arm", "Set");
                arm[0].setTargetPosition(128);
                arm[0].setPower(.25);
            }else if(loopcnt==128){
                armzeroctl.setMotorControllerDeviceMode(DcMotorController.DeviceMode.READ_ONLY);
            }else if(loopcnt>128&&(armzeroctl.getMotorControllerDeviceMode()==DcMotorController.DeviceMode.READ_ONLY)){
                telemetry.addData("Arm", String.format("Position: %d Target: %d busy: %b", arm[0].getCurrentPosition(),arm[0].getTargetPosition(),arm[0].isBusy()));

            }
            if(loopcnt<Integer.MAX_VALUE)
                ++loopcnt;
```
If I use a very small delay the motor will not move at all because the writes were never done to the controller telling it to move the motor. **The issue is most likely because any pending writes are not completed during SWITCHING_TO_READ_MODE.**.

If this is a feature and not a bug how do I wait for writes to be completed before switching to READ_ONLY mode? Could someone add a function such as pendingRW() which returns true if there are any pending reads or writes. Using a hardcoded value is not a good idea for a delay.

Are any pending writes supposed to be flushed while in SWITCHING_TO_READ_MODE? If so then this is a software bug in the FTC Application.

Leaving the encoder in WRITE_ONLY mode allows for correct behavior of the rotary encoder.

cmacfarl, 2016-11-18:
All reads and writes to motor controllers are synchronous with this season's SDK.  For this reason I believe your problem here is solved.  If you are still having trouble please post a question over on the forums.

closed 2016-11-18 by cmacfarl