ftctechnh/ftc_app #104: NXT EOPD

pjtnt11, 2015-12-10:
Is there NXT EOPD support? I can't find it in the list of sensor when I am setting up the robot for the legacy module.

magneticflux-, 2015-12-10:
Interesting! I had no idea there was an EOPD sensor for the NXT. I doubt there is a special class for it, since NXT support is lacking compared to MR sensors and such afaik. [This](http://ftcforum.usfirst.org/showthread.php?4217-What-to-do-while-we-wait-Core-Device-Interface-Module-and-sensors) might be of interest, it mentions the EOPD sensor.

pjtnt11, 2015-12-10:
@magneticflux- To be more accurate, it was a HiTechnic EOPD. Also, I figured out that you need to treat the EOPD as a light sensor in order for it to work.

cmacfarl, 2017-12-04:
Houskeeping.

closed 2017-12-04 by cmacfarl