ftctechnh/ftc_app #107: HiTechnicNxtIrSeekerSensor.getIndividualSensors throws ArrayIndexOutOfBoundsException

GorillaCoder, 2015-12-16:
HiTechnicNxtIrSeekerSensor.getIndividualSensors has a bug

The code is:
	public static final double[] DIRECTION_TO_ANGLE = new double[]{0.0D, -120.0D, -90.0D, -60.0D, -30.0D, 0.0D, 30.0D, 60.0D, 90.0D, 120.0D};
	…
	public IrSeekerIndividualSensor[] getIndividualSensors() {
	…
           for(int var2 = 0; var2 < 9; ++var2) {
               double var3 = DIRECTION_TO_ANGLE[var2 * 2 + 1];
	        ...

So, var2 ranges from 0 to 8. 8*2+1 is 17. DIRECTION_TO_ANGLE has 9 slots. 17 is bigger than 9. Boom.

KingofGamesYami, 2015-12-16:
Helpful hint, Github markdown can do java syntax highlighting.  Just use a triple ` followed by java before your code, and a triple ` at the end of your code block.

```java
public static final double[] DIRECTION_TO_ANGLE = new double[]{0.0D, -120.0D, -90.0D, -60.0D, -30.0D, 0.0D, 30.0D, 60.0D, 90.0D, 120.0D};
…
public IrSeekerIndividualSensor[] getIndividualSensors() {
…
for(int var2 = 0; var2 < 9; ++var2) {
double var3 = DIRECTION_TO_ANGLE[var2 * 2 + 1];
...
```

GorillaCoder, 2015-12-17:
Cool. Thanks. Updated.

calebsander, 2015-12-24:
I think this is due to a confusion between there being 9 zones and 5 actual internal sensors. `9` should be replaced by `5` in the `for`-loop as well as in the length of the returned array.

calebsander, 2016-02-11:
Still an issue in yesterday's release

GorillaCoder, 2020-03-19:
Please close this issue.

closed 2020-03-19 by cmacfarl