ftctechnh/ftc_app #108: gamepad1.right_stick_y is not really right_stick_y

FTC6499, 2015-12-16:
It is really right_trigger. You can easily test this for yourself by doing the following:
```loop
@Override
public void loop(){
                telemetry.addData("Joystick", String.format("Right: %f Left: %f %f %f" , gamepad1.right_stick_y, gamepad1.left_stick_y,gamepad1.right_trigger,gamepad1.left_trigger));

}
```
I do not have the same issue for gamepad2.

Note that I have the switches on the back of the controller set to 'X' as per offical FTC instruction.

magneticflux-, 2015-12-17:
@FTC6499 [We](https://github.com/Pattonville-Robotics) don't have this problem. Is it something with your particular gamepad? Any data would be useful.

KingofGamesYami, 2015-12-17:
IIRC, the mode button will cause this behavior.  Simply press it again to get the desired behavior.

[image]

The mode button is the button between the small red LED and the dpad, near the left joystick.

FTC6499, 2015-12-18:
It is not the mode button. According to http://support.logitech.com/en_my/article/21691?product=a0qi00000069ueWAAQ the mode button swaps the D-pad and the left analog stick. I can veryify that such happens when pressing it as descripted in the logitech article for the F310 which is the controller we (team 6499) are using.

**For some reason doing a factory reset resolved the issue.**

closed 2015-12-18 by FTC6499