ftctechnh/ftc_app #111: DcMotor.setChannelMode deprecated?

techgod52, 2015-12-20:
In the example OpMode `ExampleDriveWithEncoders.java`, the lines that reset the encoders have a deprecated function: `leftMotor.setChannelMode(DcMotorController.RunMode.RESET_ENCODERS);`. It says that setChannelMode is deprecated. Why is this deprecated, and what is the correct method to be using to reset the motors? Or is this not supposed to be deprecated?

techgod52, 2015-12-20:
Note: It is also being used to run the motors to position: `leftMotor.setChannelMode(DcMotorController.RunMode.RESET_ENCODERS);leftMotor.setChannelMode(DcMotorController.RunMode.RESET_ENCODERS);`

GearGrinders, 2015-12-20:
setChannelMode is deprecated, meaning that an older version had the
function go by that name. The name being used by the current version is
"setMode()". Hope that was helpful.

techgod52, 2015-12-21:
That helped, thank you!

closed 2015-12-21 by techgod52