ftctechnh/ftc_app #113: DcMotor's setPowerFloat() doesn't actually do anything

nrizzio, 2016-01-01:
We want our motors to float when power is set to zero, that is, not lock up the motors, but let them roll freely until they stop on their own. When we call `DcMotor.setPowerFloat', nothing changes.

To further test this, we set up an if statement which reads `DcMotor.getPowerFloat()`, and if it is false, then call `setPowerFloat`.  Again, nothing changes.
```java
// Inside loop function
if(!motor.getPowerFloat)
{
    motor.setPowerFloat();
}
// When motor.getPowerFloat() is called here, it is still false

// Sometime later
motor.setPower(0); // Here the motor locks up and stops.
```

calebsander, 2016-01-01:
Unable to reproduce with the Modern Robotics motor controllers. Calling `setPowerFloat()` lets the motor rotate much more freely and causes `getPowerFloat()` to return `true` soon afterwards. Here is my test code:
````java
public class MotorFloatTest extends OpMode {
	private DcMotor motor;

	public void init() {
		this.motor = this.hardwareMap.dcMotor.iterator().next();
	}
	public void loop() {
		if (this.gamepad1.a) this.motor.setPowerFloat();
		if (this.gamepad1.b) this.motor.setPower(0.0);
		this.telemetry.addData("Float mode", this.motor.getPowerFloat());
	}
}
````

magneticflux-, 2016-01-01:
@nrizzio If you're experiencing weird issues, check battery, wiring, and your app version. It's entirely possible that any one of those caused ```setPowerFloat();``` to fail.

nrizzio, 2016-01-02:
@calebsander @magneticflux- Thanks for the info, I'm going to close the issue until I can narrow it down further. From the looks of it, it's probably an issue on our end.

Thanks again!

closed 2016-01-02 by nrizzio

nrizzio, 2016-01-02:
I figured out what was happening: calling `setPower(power)`, regardless of the value of `power`, will reset the motor float condition to false. Not sure if this is intentional or not, but at least I know what to work around.

@calebsander Thanks for the example code - it really helped.

calebsander, 2016-01-02:
Yes, the way it is internally implemented is that calling `setPowerFloat()` will send a special value (representing the desire to let the motor float) to the same "register" on the motor controller that stores the desired power for the motor. When you call `setPower()` it will send a new value to that register that represents an actual power level, effectively overwriting the float mode.

magneticflux-, 2016-01-02:
The curse of no documentation strikes again! Downloading a [decompiler](http://jd.benow.ca/) would be a good idea so you have a better understanding of what is going on behind the scenes. I know it helped me figure out why what I was doing with encoders wasn't working.

calebsander, 2016-01-02:
Android Studio actually has a built-in decompiler, so you can right-click on `setPowerFloat` and select `Go To > Declaration` or `Go To > Implementation(s)` and it will automatically decompile the necessary class file.

magneticflux-, 2016-01-02:
@calebsander I've found that decompiler to be lackluster. We're forced to use an out-of-date version of Android Studio by our school district though, so it might be better in a newer update.