ftctechnh/ftc_app #119: Whilst in a competition...

knightsean00, 2016-01-10:
When you're in a competition, and there is a lot of people, then the WiFi direct will not work the way it is supposed to, and so, as a result, you have to go where there is no cell saturation, and go back inside connected and driving your robot. However, if you do not have the luxury of doing that as you're just about to compete, it is very frustrating that the app will restart the phone while the robot is on field, or the phones just won't connect.

One time, no robots moved on the field in one match, and it was very sad.

magneticflux-, 2016-01-10:
@knightsean00 It is possible to select the Wifi Direct channel manually. You can download a scanner to find the sparcest channel and use that one. There are 16 I believe.

nrizzio, 2016-01-10:
The issue tracker is for issues relating directly to programming; as this issue is caused by WiFi interference and not source code, you would benefit more from going to the [ZTE Phone Configuration and Setup forum](http://ftcforum.usfirst.org/forumdisplay.php?161-ZTE-Phone-Configuration-and-Setup) on FIRST's website and posting your issue there.

ftctechnh, 2016-01-10:
Hi Sean,

I'm sorry that you had an issue at your event, but the WiFi Direct system has been tested pretty thoroughly and you should be able to get reliable performance, even if there is a fair amount of wireless activity in the venue.   We've run tests where we had 59 or 60 robot pairs running simultaneously in a very small venue.  The ping times for these high volume stress tests were low (on the order of 5 msec or lower).  There is some planning that the event host/FTA can do to help mitigate the risk of wireless interference, but in general, the WiFi Direct performance of the ZTE Speed phones is very good.

If you are having issues, as others suggested, please visit the forum to see if the community can help resolve your issues.

closed 2016-01-10 by ftctechnh