ftctechnh/ftc_app #120: Contribution Guidelines

lukethecoder64, 2016-01-12:
As you may know, there are many seasoned programmers (including me) who would like to help make this SDK better. This cannot be done without some basic contribution guidelines. FIRST, please leave an official opinion on this issue.

magneticflux-, 2016-01-12:
@lukethecoder64 So far they have been hostile to contributions. The main code base is obfuscated, and they have numerous posts saying that it is not possible to release source code due to various copyright/licensing issues.

lukethecoder64, 2016-01-12:
I am well aware of the source hogging. However, contributions can still be made. If they are against community help, could they say so in the readme?

YoshiRulz, 2016-01-12:
As this is a symptom of the underlying licensing problems, perhaps the new software and driving systems should be based entirely off open source code? That goes without saying, many veteran teams' software departments would be more than happy to contribute, check for flaws/exploits, maybe run test matches etc.

magneticflux-, 2016-01-13:
@YoshiRulz I highly doubt that they will scrap all current progress on the system. What they have currently is incredibly complex and was a huge undertaking to create. It seems more likely that they will try to press through the licensing issues.

xyzzycoder, 2016-01-13:
I think that they should have gone with Arduino.

Aron

lukethecoder64, 2016-01-13:
I absolutely agree they should have gone with arduino, @xyzzycoder. However back to the OP: FIRST, please follow GitHub etiquette and have contribution guidelines.

PranavMathur, 2016-01-13:
@lukethecoder64 I'm not sure whether or not FIRST will agree to having a set of contributor guidelines, but I wholeheartedly agree with the sentiment that they should follow proper git style. Commits should not be made only for major releases, and there should be frequent commits with descriptive messages. Pull requests should also be considered, as some contain code that is useful to every team.

lukethecoder64, 2016-01-30:
*Sigh*
Oh well. I guess I will ask on the forums.

P.S. Does anyone else find it weird that FTC will not address issues related to open source, such as this one?

Ernest314, 2016-02-03:
Not much I can add that hasn't been said before... It really is very frustrating how cryptic FIRST is being.

cmacfarl, 2016-11-16:
Contribution guidelines are at .github/CONTRIBUTING.md.   Further discussion on this topic would likely be better addressed over on the forums where there are a wider variety of FIRST staffers and volunteers participating.

closed 2016-11-16 by cmacfarl