ftctechnh/ftc_app #122: Fatal Errors on pre-set commands

FTCTeam4081, 2016-01-15:
So when we last downloaded the App master and imported it into Android Studio we got a bunch of fatal errors. They were all on the pre-set methods that control motors. 
![fatal error
[image]
](https://cloud.githubusercontent.com/assets/13933347/12364730/693e8384-bb9e-11e5-9873-46f141af4a65.png)

The photos show some examples. We tried going back and using an older version, which worked fine. Anybody have any ideas what's going on?

smh3005, 2016-01-15:
Have you imported everything correctly?

Did you install the SDK in the right folder?

calebsander, 2016-01-16:
If rebuilding doesn't work, this is always fixed for me by going to `File > Invalidate Caches/Restart`.

FTCTeam4081, 2016-01-29:
@smh3005, yes. We imported it the same way we did the older version.
@coder-bot, we tried that, and this is what we got:
[image]
We tried it twice.
@calebsander, we tried that too, and we got the same errors.
We've tried this on a few different computers, and we got fatal errors on all of them.
Thanks for all the help!

magneticflux-, 2016-01-29:
Can you walk through how you imported the project? It seems as if your libraries are not set up properly.

rgatkinson, 2016-09-20:
It's likely that any underlying issues here are resolved in the current (2.2) release.

closed 2016-09-20 by rgatkinson