ftctechnh/ftc_app #125: Driver Station robot status display can appear greyed-out despite being ready to run

calebsander, 2016-01-16:
This shows what I am describing (6055-C-RC is being used as a driver station):
[image]
Pressing `Init` initializes the OpMode as normal, and joystick and telemetry information work fine, however the robot controller status still appears darkened. This seems to happen every time in the the `16.01.04` release when we re-run the robot controller app from wireless ADB without closing it or the driver station app. The ping time continues to update itself, and I assume that the battery levels do as well, but it is confusing to drivers when it looks like the driver station is failing to connect to the robot controller.

magneticflux-, 2016-01-16:
@calebsander Team 2866 has also experienced issues running the robot controller using Wireless ADB. Restarting fixes it, but is time consuming and defeats much of the purpose of Wireless ADB.

rgatkinson, 2016-01-16:
@magneticflux- in your use of Wireless ADB, which wireless network, where you seem these issues, are you using to connect to the robot controller: then Wifi-Direct one that it owns, or a second (traditional) network?

calebsander, 2016-01-16:
For us, everything seems to work fine with the exception of the bug in the driver station app that was just introduced in the beta release. The only temporary fix for it seems to be to restart the driver station app.

magneticflux-, 2016-01-16:
@rgatkinson We have an access point in our room that both the robot controller and the laptop are connected to.

calebsander, 2016-01-16:
@magneticflux- We have been doing it the same way with no issues.

magneticflux-, 2016-01-16:
@calebsander It seems that there is some sort of desynchronization between the robot controller and driver station. The driver station claims that it is connected, but pressing Init or Start doesn't do anything.

xyzzycoder, 2016-01-16:
> 
Could anyone share a link that highlights debugging "how-to" on the new platform? Specifically, with a debugger (not just logging)?

Thank you,

Aaron

calebsander, 2016-01-16:
@xyzzycoder I would recommend you post on the [Technology Forum](http://ftcforum.usfirst.org/forumdisplay.php?156-FTC-Technology) instead. GitHub issues is intended for bug reports instead of help requests.

calebsander, 2016-02-20:
Still seems to happen in 1.6

xyzzycoder, 2016-02-21:
I don't think I posted here. 

Aaron

cmacfarl, 2016-08-20:
@calebsander , can you please retest with SDK 2.0 (currently in the beta branch) and close this if you no longer see the issue?  Thanks.

calebsander, 2016-08-24:
Seems fixed. I will reopen if I manage to reproduce it this season.

closed 2016-08-24 by calebsander