ftctechnh/ftc_app #126: Inconsistency between OpMode methods — some are abstract and some are empty

calebsander, 2016-01-17:
`OpMode` defines `init()` and `loop()` as abstract methods while `init_loop()` and `start()` and `stop()` simply have no body. There are certainly cases in which one would want to write a non-abstract class that inherits from `OpMode` without either an `init()` method or a `loop()` method, so it seems they should not be declared abstract. This change shouldn't break any existing code.
````java
public abstract void init();
public void init_loop() {}
public void start() {}
public abstract void loop();
public void stop() {}
````

MeRPG, 2016-01-17:
The intention is to require use of these methods. Anything not requiring those methods isn't really much of a program for a robot. If you really wanted to, you could make your own abstract class extend OpMode and override init and loop with empty bodies.

calebsander, 2016-01-18:
We have several test programs that omit one of those methods for any number of reasons. It just seems a little strange that those methods are required to be implemented when they could easily default to empty.

calebsander, 2016-02-11:
Still an issue in yesterday's release

EtherTyper, 2016-12-01:
@calebsander Would a local non-abstract subclass implementing those methods work for you?

```java
package org.firstinspires.ftc.teamcode;

public class OpMode extends com.qualcomm.robotcore.eventloop.opmode.OpMode {
    @Override
    public void init() {
        
    }

    @Override
    public void loop() {
        
    }
}
```

calebsander, 2016-12-02:
We already have our own `OpMode`-extending class that implements all the methods for us. I know you can make empty implementations of the methods, but it seems strange that the SDK doesn't support it natively. Also, I think it is preferable to keep `OpMode` abstract because you don't want to actually register it.

EtherTyper, 2016-12-02:
@calebsanders Good point, and I totally agree that they should have a dedicated test class in the standard library.

olekmali, 2016-12-02:
You don't have to do that. You can have a class that extend an OpModa 
and label it as abstract, and then extend it into several more classes. 
For example see 
https://github.com/MatthewMateusz/ftc9985-2016/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/PushBotAutomation.java 

Olek.

Dr. Aleksander 'Olek' Malinowski
phone: +1 (309) 306 1131
skype: olekmali
http://www.bradley.edu/academic/departments/electrical/faculty/profile.dot?id=10917dd0-3092-4e27-95f5-7eb07a52e1c5

calebsander, 2016-12-02:
Yes, that is what I said. We have an abstract class called `BaseOpMode` which extends `OpMode` and implements each stage method by running some utility code and then invoking an overridable empty method. `BaseOpMode` would be a valid `OpMode` in its own right, but we declare it `abstract` in order to prevent it from being registered.

cmacfarl, 2017-12-04:
Rather easy workarounds exist.  Closing while doing housekeeping.

closed 2017-12-04 by cmacfarl