ftctechnh/ftc_app #132: OpMode telemetry can be displayed before the OpMode starts

calebsander, 2016-01-24:
Steps to reproduce:

1. Initialize and start an OpMode that writes to telemetry during `loop()` only
2. (Optional) Stop the OpMode
3. Select another OpMode from the menu
4. Select the first OpMode again
5. Initialize it
6. Notice that the last telemetry data appears on the screen

This can be confusing because it appears as though the OpMode is running yet the telemetry data is from a previous run and doesn't update.

Screenshot:
[image]
Here, `BNO055` is an OpMode that only displays that information during `loop()`, so the bottom of the screen should be empty during the `init()` phase.

lukethecoder64, 2016-01-24:
Ooh, good catch!

Luke
     __
    /   /
   /   /
  /   /
 /   /______
/_________/

calebsander, 2016-08-24:
Seems fixed in 2.0.

closed 2016-08-24 by calebsander