ftctechnh/ftc_app #135: Core Device Interface Module LEDs do not work

nitrocalcite, 2016-01-27:
While using the .setLED method on the CDIM, the LEDs do not light up, or shut down.  The documentation on the method is horrible as well - I'm not sure which integers represent what color LEDs.  I've tried 0,1,2, but none of these have any effect on the light state.  The method is called during init, not the main opmode activity, but this should be irrelevant.  I have not tested it during main activity.

calebsander, 2016-01-28:
I'm unable to reproduce. This is the program I have used to test:
````java
public class DIMLightTest extends OpMode {
	private DeviceInterfaceModule dim;

	public void init() {
		this.dim = this.hardwareMap.deviceInterfaceModule.iterator().next();
	}
	public void loop() {
		this.dim.setLED(0, this.gamepad1.x); //blue
		this.dim.setLED(1, this.gamepad1.b); //red
	}
}
````

calebsander, 2016-01-31:
Could you attach your code so we can take a look at it?

nitrocalcite, 2016-01-31:
Sorry, I've been meaning to test your code but our robot is in a very large number of pieces right now, and will remain that way until middle of this week.  I'll either test it then or try to make a test rig.

TeamTorch5942, 2016-02-03:
While the documentation on the CDIM is terrible, we use the LEDs in competition successfully. If you use the code provided by @calebsander be sure to name your CDIM 'dim' in the robot config. 

We have these functions to toggle the LEDs. Blue is 0 and Red is 1. 
```
    public void blueLED (boolean state){dim.setLED(0, state);} 
    public void redLED  (boolean state){dim.setLED(1, state);} 
```

calebsander, 2016-02-03:
It doesn't actually matter what you name the device in the configuration file; my code will work as long as at least one is declared in it. Note that I don't fetch the DIM by name.

nitrocalcite, 2016-02-12:
Closing since this seems to be an issue on our end.

closed 2016-02-12 by nitrocalcite