ftctechnh/ftc_app #136: It is misleading to have HiTechnicNxtServoController.getServoPosition(int channel) return 0

calebsander, 2016-01-31:
If this is not supported, it should throw an error rather than silently return 0.0, which looks like a legal value.

lukethecoder64, 2016-01-31:
Yes, but doesn't that mean that every time we read a servo we have to use try/catch?

calebsander, 2016-01-31:
Not if you made it throw a `RuntimeException`. If you forgot that it was on a HiTechnic controller, then it would throw the error once and you would change the code to stop calling it. If it was on a Modern Robotics controller, it wouldn't ever throw an exception.

magneticflux-, 2016-01-31:
The correct exception to be thrown, inline with default libraries and the [gyro](https://htmlpreview.github.io/?https://raw.githubusercontent.com/ftctechnh/ftc_app/master/doc/javadoc/com/qualcomm/robotcore/hardware/GyroSensor.html#rawZ()) code, would be ```UnsupportedOperationException```, which is a ```RuntimeException```.

calebsander, 2016-01-31:
Yes, I meant that throwing any sort of `RuntimeException` would be sufficient to not require a try-catch statement every time the method is called.

lukethecoder64, 2016-01-31:
Thanks for the clarification, @calebsander and @magneticflux- . Yes, I agree that an exception should be thrown. +1.

calebsander, 2016-02-06:
There is a similar issue with `getPwmStatus()`. It always returns `PwmStatus.DISABLED`.

calebsander, 2016-02-11:
Still an issue in yesterday's release

calebsander, 2016-08-20:
Seems fixed in 2.0.

closed 2016-08-20 by calebsander