ftctechnh/ftc_app #137: Continuous Rotation servos should be treated differently from normal servos

calebsander, 2016-02-01:
There are several points where I think the SDK would benefit from having a clear split between CR and positional servos:
- They should use `setPower()` with `DcMotor`-like values instead of `setPosition()` to make their usage more intuitive
- When no `OpMode` is running, they should be stopped before PWM is disabled, because otherwise they can do serious damage and disqualify teams
- They should not be included in the `HardwareMap.servo` `DeviceMapping` because iterating over it and doing the same thing for each servo would cause CR servos and normal servos to act differently
- To these ends, this distinction should exist in the configuration file so it is harder for a team to confuse the two sorts of devices

JacobAMason, 2016-02-01:
I think it'd be good to make the two distinct as well. We came across the same thing and I know this would be a relatively easy fix. Until one is committed, it's not too hard to write a wrapper. We plan on doing that soon, so when we do, I'll edit this comment and link it here.

lukethecoder64, 2016-02-01:
+1. This should be obvious.

Luke
     __
    /   /
   /   /
  /   /
 /   /______
/_________/

JacobAMason, 2016-02-03:
Here's an easy sample FTC3486/FTCRC_Extensions@62491e2006eb88b22fa8f0f719cf28666afa8c65

calebsander, 2016-08-20:
Fixed in 2.0.

closed 2016-08-20 by calebsander