ftctechnh/ftc_app #138: HardwareMap should be a single DeviceMapping and require a cast when getting control components from it

calebsander, 2016-02-01:
Having a separate `DeviceMapping` object for every conceivable sort of device seems undesirable because it makes the `HardwareMap` class harder to maintain and results in somewhat different sensors (such as the HiTechnic gyro and the MR gyro) being considered the same. To solve this, I propose the following:
- `HardwareMap` should internally store a `Map<String,HardwareDevice>` of every device specified in the config file
- As #95 proposed, the sensor classes should be interfaces. They should specify a *lowest common denominator* set of methods instead of forcing implementing classes to throw `RuntimeException`s when the method is unsupported. This would also allow for the `HiTechnicNxtSensor...` classes, the `ModernRoboticsUsb...` classes, etc. to reuse code, which would make their implementations much more elegant.
- When getting a device, it could be cast to its actual class or, in the case of sensors, the interface it implements. Trying to declare a sensor as the wrong type will immediately throw an exception showing what class the sensor is in the configuration file and what the user was trying to declare it as.

Green0Photon, 2016-02-01:
I agree, the way `HardwareMap` is currently set up make it a pain to write the generator code for my high level wrapper for my team.

The only issue I see is with 1. What happens when two hardware names clash in the config? I suppose this issue might be resolved the same as the intrahardware naming code, which I haven't tested to see what happens when it clashes.

calebsander, 2016-02-01:
I am fairly certain that it actually isn't possible to save a configuration file with clashing names, even if the devices are different.

calebsander, 2016-08-20:
Fixed in 2.0.

closed 2016-08-20 by calebsander