ftctechnh/ftc_app #143: Make time unit consistent

calebsander, 2016-02-06:
These are all the locations I see time units being used:
-`OpMode.time` (s)
-`OpMode.getRuntime()` (s)
-`GyroSensor.getRotation()` (deg/s)
-`ElapsedTime.time()` (s by default)
-`LinearOpMode.sleep()` (ms)
-`Gamepad.timestamp` (ms)
-`PWMOutput.setPulseWidth*()` (ms)
It seems like these should all use the same unit in get and set methods, regardless of the unit they actually use to store the values. Seeing as `second` is a base SI unit and most of less obscure methods seem to use it, I would argue that it is a better choice.

rgatkinson, 2016-02-06:
By contrast, most of the OS timing values that are in an implicit unit are in either ms or in ms + ns. Some, however, have an explicit TimeUnit qualifier.

calebsander, 2016-02-06:
I don't think the choice of prefix matters all that much; it is more important (especially with very uninformative documentation) to have the unit be consistent. Beginners should not have to know what unit each method or variable is using.