ftctechnh/ftc_app #150: Driver Station losing controllers

vwestin, 2016-02-19:
Working with the 16.02 beta, MANY things have greatly improved!

When we have been testing, our driver station (app from the doc/apk folder) sometimes just loses the ability to see the gamepads. Restarting the app will sometimes fix it, but other times we have to reboot the phone. We are seeing this with 2 different driver station phones, so it does not appear to be a hardware specific issue. We have also replaced to OTG cable, but still see this happening. We can go 30+ minutes between issues, but sometimes it happens in only a few minutes.

We also are seeing the drive station just ending communication with the robot. The robot still shows the robot running the OpMode, but no driver communication.

We have also seen the Driver Station app just lock up entirely (no response to menu selection or other touches). Killing it was not enough to return operation, but rebooting the phone worked.

Not sure if anyone else is seeing this sort of Driver Station challenges.

Vince Westin
Coach, #5940 Rockin' Robots
vincewestin@gmail.com

skeltonn, 2016-03-03:
We have been preparing for the MO State qualifier and have been encountering this issue. Just thought I would confirm. Our other team was using a set of controllers that was working with their phone, but were unrecognizable on ours, so I agree it is unlikely that the hardware of the controllers or the phone is causing the issue.

Pattonville Robotics Teams 2866 & 2867

slylockfox, 2016-03-04:
We started noticing this issue in the last week as well, with 16.02.  Cyber Ducks 8862

vwestin, 2016-03-06:
After further testing, we are convinced that this is a problem in the protocol with the robot controller. Only when the robot controller gets into a funny state are we having trouble with the driver station. We had a recent scrimmage where we had a phone/robot pair (simple demo bot) running for most of 2 hours without a single hard reboot on either phone - even with a change in robot battery. It does appear that the protocol (state machine?) can get caught up in the communications and only a hard reset of both phones has given us a clean fix.

rgatkinson, 2016-03-06:
Might the logs from the driver station or the robot controller (ideally both) be available?

jumbojet0105, 2016-03-16:
We have had a very similar issue.  Today, we tried using a different USB hub, and that seemed to greatly help.  Have you tried replacing anything besides the OTG cable?  For us it seems to be just a wiggly connection where the wire enters the USB hub.  However, we are also going to try some code-related things.  Never seen the Driver Station lock up, however, so possibly they are two separate issues.
Programmer, 9968 Crossfire

ajtrowell, 2016-09-17:
Our team had intermittent controller drop outs which ceased when we replaced the usb hub.

rgatkinson, 2016-09-20:
Has anyone recently observed this issue such that replacing the USB hub would not resolve it? There was much work on the protocol during the summer; I suspect that that likely resolved the software issues.

2000andrewbowen, 2016-12-15:
We have recently seen this problem. We could not connect the controllers or doing anything on the driver station app. This as happened twice, once during a match and once more during a practice. At the practice we tried just one controller and no hub but it still happened. The only way to fix it was to restart the phone.

KingofGamesYami, 2016-12-15:
Our team has observed several categories of gamepad disconnects / malfunctions.  For reference, we currently use SDK 2.4 on ZTE Speed Phones (both RC and DS).  Some of the behavior described may have been exhibited with SDK 2.35 but not yet encountered on 2.4.

1) DS refuses to recognize second gamepad -- The DS will connect one gamepad, but not both.  We've tried swapping the gamepads but all gamepads involved are able to connect via the OTG cable - just not with the USB Hub (KOP monoprice, andymark 2:1 y, and a freebie from a couple years ago).  This is usually resolved by some magic voodoo of unplugging the OTG cable and restarting the app, though the fix does not always work on the first try.
2) DS refuses to recognize any gamepad connected to a USB Hub -- The DS will recognize a gamepad connected to an OTG cable, but the same gamepad will not show up connected to any port on the USB Hub.  We did try swapping the USB Hub, but did not have the Andymark Y cable at the time.  The fix is the same as 1.
3) DS randomly drops connection to both gamepads -- Sometimes, while testing, our robot will stop moving and it's because our gamepads have both disconnected.  We just have to do start+(a/b) again to resolve this issue.
4) Gamepad will return garbage values (ei trigger is half pressed when it's not at all) -- This doesn't happen very often, and I recall it occurring with the NXT control system as well.  I doubt it's fixable and doesn't occur often enough to warrant a fix.  Our driving team has been informed of the issue and the solution is to use whatever value is supposedly wrong (ei pushing the trigger).

SpencerSharp, 2016-12-15:
I thought nobody else had this problem! I've had all four of the listed problems very often, and even replacing all of the cables and the controller does not fix it. It's a massive pain and costs us a lot of time both in and out of matches.

I actually get the garbage value issue a lot though, and I think it is certainly warranted to fix. I'm not sure if it is the fault of the devs though. It is almost always with the triggers, I'm not sure if I've encountered it outside of the triggers.

KingofGamesYami, 2016-12-15:
We've only encountered it on "analogue" values such as the joysticks and triggers.  Boolean values (a, b, etc.) seem to be unaffected.  I'm fairly certain this is an issue with the F310 gamepad.

Scroogeee, 2018-04-08:
Happening in 3.7: Phone disconnects from the gamepads, changing OTG cable only temporarily resolves the issue but putting in the (so-thought) "broken" cable back in works again sometimes but in the worst case: After a restart of the whole phone no cable works anymore... After a couple of restarts it works again... I have no idea how to resolve this