ftctechnh/ftc_app #151: Robot Controller Hangs

vwestin, 2016-02-19:
The new 16.02 has fixed lots of issues. The dynamic reconnect when devices are lost/recovered is wonderful!

We are seeing two challenges with the Robot Controller now. One is that the program appears to occasionally enter an endless loop (dies with the screen error 'Unfortunately, FTC Robot Controller has stopped.' after a several minute pause of no response to any touch). We do not see a way to get a diagnostic log for these crashes.

The other is that sometimes it simply stops running our loop. The screen says the OpMode is still running. The logs show that our debug messages for the last loop of our code ended normally. But then the log just shows no further attempts to run our code. Some of the occasional timeout messages, but no further communication.

If there was a way to have the logs run longer it might allow us to see more of what is happening in our code. Sometimes when we disconnect the phone for diagnostics, half of the log space is taken up with messages about tearing down the robot services.

gearsincorg, 2016-02-19:
I'm not pointing at this as a problem, but I'm interested to know if these hang-ups tend to occur in situations where static discharge may be occurring.

These would be:  
On contact between the robot frame and the metal field perimeter.
On contact with the exposed metal of the Mountain (not the ramp itself).
On contact with another robot.

If the occurrence is "location specific" is points to an external influence.

Phil.

nrizzio, 2016-02-19:
> The other is that sometimes it simply stops running our loop. The screen says the OpMode is still running. The logs show that our debug messages for the last loop of our code ended normally. But [...]

Related to SwerveRobotics/ftc_app#77?

vwestin, 2016-02-20:
We had a lot of problems with static with the prior code - it would kill us dead. We recover from most of those now.

This problem can happen in the middle of the field, touching nothing but debris. The strangest have been when they are working an issue with the Driver Station, so they send a remote Reboot Robot - and the Robot Controller screen freezes on that until Android reports that the app has stopped.

So this does not appear to have anything to do with static. Somehow the app is getting into a tight loop and not coming out.

We have not been seeing the random OpMode not starting as described in #77. The OpMode starts fine, and runs many iterations of the loop - and then just stops.

We have not jailbroken the phone to be able to browse all of the phone directories. Is there another way to easily see all the log file names to be able to grab them? We would be glad to attach any logs we can find.

gearsincorg, 2016-02-22:
Hi.  Jailbreaking is not required to see the standard logs.  For windows, use phone "settings/Connect to PC" to select "Media".  This will make the phone appear as a drive to the computer.
It will show up as the Phone model.  Open that folder and the logs are in the root folder.  Scroll down.

nitrocalcite, 2016-03-03:
We eventually solved a similar issue by reloading the .aars.  Maybe you can check the checksums (heh) of the of your .aars against a fresh copy?

vwestin, 2016-03-16:
Our robot was using 7 modules on the USB ports of the Core Power Distribution Module - 4 Motor Controllers (8 active motors), 1 Legacy Module, 1 Servo Controller, and 1 Device Interface Module. Watching the robot run, we found that we were having the issues most frequently with many (sometimes all 8) motors active. We also noted that the reported main battery voltage would drop below 12V during heavy motor usage. And it was during those times were were most likely to see the loss of connection to modules and have the robot stop.

We decided that the problem might be that the 5V power for the USB bus was over extended. We dropped back to 2 of the new Motor Controllers and 1 Legacy Module (based on the way that our robot was wired, this seemed the best option to minimal USB usage). We used 2 legacy motor controllers and a legacy servo controller through the Legacy Module to maintain our motors/servos, and we gave up on LED control (we just powered the LEDs directly from the 12V out of the PDM).

Since making that change, we have not had a single issue with a hang on the robot controller or the driver station. So it appears that when the power drain on the system is high, the USB power is not staying clean.

gearsincorg, 2016-03-16:
Hi

You mention that you "gave up on LED control".  What exactly were you doing with the LEDs and how were you controlling them.  Is it possible that the LED's were contributing to the problem (by either placing demands on the USB traffic or by generating significant radiant noise).

Did you also reduce the number of motors, of did you just gang them up?  A reduction in motors would also have reduced the 12V sag.

vwestin, 2016-03-16:
We have the motors on the legacy controllers. Without all 8, we cannot run. And each needs separate control.

We had already unplugged the LED lights and the relay block we were using to control them. That made no difference. We can add or remove them in the new configuration without seeing the controller issues. We had also tried removing just the Device Interface Module, to see if that was part of the problem. We did not notice a change in the frequency of incidents with or without it.

TeamTorch5942, 2016-04-15:
We're having this issue as well. It generally happens several times a day. We run 8 motors, 6 servos, 1 gyro, 1 IR distance sensor, and 1 touch sensor. Generally the app freezes when we run into our mountain. We haven't determined yet if it's the shock from hitting it too hard or static discharge, though these comments make it look like it may be the latter. Is there any way to prevent this issue? We really can't afford to have the app freezing on the field at Worlds. We're already using anti static spray on our wheels which has stopped us from disconnecting in the past, but either the latest update or adding 2 more motors to our robot has made the app start to freeze.

gearsincorg, 2016-09-04:
Closed, as this issue relates to previous season code & hardware.
Current season code is better at detecting endless loop situations.
Teams are encourages to obtain firmware updates on their MR hardware to mitigate possible reset conditions.
Any new/similar issues this season should be started on a new issue.

closed 2016-09-04 by gearsincorg